SerBrynden/PiClock #3: Tomorrow.io gives completely wrong pressure

feh123, 2023-04-16:
Hi @SerBrynden I seem to remember a comment about this when we were discussing the loss of Dark Sky. I have been using OWM for a few weeks and I thought I would switch to Tomorrow to see if the forecasts were better. Not that OWM was poor, in fact it seemed quite good. I have an API from 2021 for Tomorrow which seemed to work just fine. However the pressure reading was 786 against a correct reading of 1024. Is there an update that fixes this? Thanks.

SerBrynden, 2023-04-16:
A few things... OWM only provides pressure at sea level and PiClock displays it in either millibars (metric) or inches of mercury, inHg, (imperial).  Tomorrow.io can provide pressure as sea level or the surface (local elevation above sea level) and PiClock displays it in either millimeters of mercury, mmHg, (metric) or inHg (imperial).  Currently, PiClock uses pressure at surface level from Tomorrow.io.  I don't know who made that decision, but looking into it more, perhaps that should be changed to sea level because it seems like most other weather providers (local news, Weather Channel, etc.) show pressure at sea level as the standard.  Would you agree with that change?  I'm also thinking about changing the pressure metric units to millibars for Tomorrow.io

feh123, 2023-04-17:
Thanks for this. Oops how stupid of me - Tomorrow was reporting the pressure in mmHg! I thought it was a faulty formula to get the pressure to mB.
I see your new files to correct this and fix other bugs. That's great. To update is there a way to download the files and just make the changes on my PiClock?
Thanks again.

SerBrynden, 2023-04-17:
Try https://github.com/SerBrynden/PiClock/blob/master/Documentation/Install.md#updating-to-newerupdated-versions

closed 2023-04-17 by SerBrynden

feh123, 2023-04-17:
Hi @SerBrynden thanks for the update commands - they worked brilliantly. Tomorrow.io pressure now mbar and I do agree about making this a sea level reading (sorry for not answering this before). 
Everything looks very good. Now to see how Tomorrow's forecasts work for the UK!