SerBrynden/PiClock #4: list index out of range error

joeysantana3, 2023-05-04:
Hello,

I just followed the setup instructions and tried running this project on my pi. I'm using Mapbox and Tomorrow.io as sources. I got API keys for both and put them in the `ApiKeys.py` file. I changed `Config.py` to match my coordinates and tried starting the app via `bash startup.sh -n -s`. I got the below error.
```
PiClock/Clock/PyQtPiClock.py", line 1049, in wxfinished_tm_daily
    f = wxdata3['data']['timelines'][0]['intervals'][i - 3 + ioff]
IndexError: list index out of range
```

Here is the full output (minus private info)
```
PiClock $ bash startup.sh -n -s
Disabling screen blanking....
Setting sound to max (assuming Monitor Tv controls volume)....
Checking for NeoPixels Ambilight...
Checking for GPIO Buttons...
Checking for Temperature Sensors...
Starting PiClock.... logging to screen.
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
map base url for radar1: https://api.mapbox.com/styles/v1/bcurley/cj712peyz0bwr2sqfndbggupb/static/-XXX.XXXXXX,XX.XXXXXX,6,0,0/100x97?access_token=p<mapbox_token>&attribution=false&logo=false
map overlay url for radar1: https://api.mapbox.com/styles/v1/bcurley/cj712r01c0bw62rm9isme3j8c/static/-XXX.XXXXXX,XX.XXXXXX,6,0,0/100x97?access_token=p<mapbox_token>
map base url for radar2: https://api.mapbox.com/styles/v1/bcurley/cj712peyz0bwr2sqfndbggupb/static/-XXX.XXXXXX,XX.XXXXXX,10,0,0/100x97?access_token=p<mapbox_token>&attribution=false&logo=false
map overlay url for radar2: https://api.mapbox.com/styles/v1/bcurley/cj712r01c0bw62rm9isme3j8c/static/-XXX.XXXXXX,XX.XXXXXX,10,0,0/100x97?access_token=p<mapbox_token>
map base url for radar3: https://api.mapbox.com/styles/v1/bcurley/cj712peyz0bwr2sqfndbggupb/static/-XXX.XXXXXX,XX.XXXXXX,6,0,0/233x248?access_token=p<mapbox_token>&attribution=false&logo=false
map overlay url for radar3: https://api.mapbox.com/styles/v1/bcurley/cj712r01c0bw62rm9isme3j8c/static/-XXX.XXXXXX,XX.XXXXXX,6,0,0/233x248?access_token=p<mapbox_token>
map base url for radar4: https://api.mapbox.com/styles/v1/bcurley/cj712peyz0bwr2sqfndbggupb/static/-XXX.XXXXXX,XX.XXXXXX,10,0,0/233x248?access_token=p<mapbox_token>&attribution=false&logo=false
map overlay url for radar4: https://api.mapbox.com/styles/v1/bcurley/cj712r01c0bw62rm9isme3j8c/static/-XXX.XXXXXX,XX.XXXXXX,10,0,0/233x248?access_token=p<mapbox_token>
libEGL warning: DRI2: failed to authenticate
libpng warning: iCCP: known incorrect sRGB profile
getting Tomorrow.io current conditions: https://api.tomorrow.io/v4/timelines?timesteps=current&apikey=<tomorrow.io_token>&location=XX.XXXXXX,-XXX.XXXXXX&units=imperial&fields=temperature,weatherCode,temperatureApparent,humidity,windSpeed,windDirection,windGust,pressureSeaLevel,precipitationType
getting Tomorrow.io hourly forecast: https://api.tomorrow.io/v4/timelines?timesteps=1h&apikey=<tomorrow.io_token>&location=XX.XXXXXX,-XXX.XXXXXX&units=imperial&fields=temperature,precipitationIntensity,precipitationType,precipitationProbability,weatherCode
getting Tomorrow.io daily forecast: https://api.tomorrow.io/v4/timelines?timesteps=1d&apikey=<tomorrow.io_token>&location=XX.XXXXXX,-XXX.XXXXXX&units=imperial&fields=temperature,precipitationIntensity,precipitationType,precipitationProbability,weatherCode,temperatureMax,temperatureMin
wxstart for radar1
wxstart for radar2
Traceback (most recent call last):
  File "/home/joey/PiClock/Clock/PyQtPiClock.py", line 1049, in wxfinished_tm_daily
    f = wxdata3['data']['timelines'][0]['intervals'][i - 3 + ioff]
IndexError: list index out of range
startup.sh: line 115:  1609 Aborted                 python3 -u PyQtPiClock.py

```

feh123, 2023-06-08:
I get something very similar. Luckily for me PiClock works (using tomorrow.io for forecasts). So it's not a fatal fault. But overnight it very often crashes with the error:
Traceback (most recent call last):
  File "/home/pi/PiClock/Clock/PyQtPiClock.py", line 1049, in wxfinished_tm_daily
    f = wxdata3['data']['timelines'][0]['intervals'][i - 3 + ioff]
IndexError: list index out of range
Aborted
Rather similar to you error. Using sh startup.sh -n -s restarts PiClock without a problem.

SerBrynden, 2023-06-21:
Fixed with commit e2b29e4

closed 2023-06-21 by SerBrynden