SerBrynden/PiClock #5: Radar of Western Europe looks different to Rainviewer

feh123, 2023-06-26:
It's not a major issue but it seems persistent and I thought I would mention it.

I have two radar maps - one at high zoom covering eastern England centred on Cambridge and a second map that covers England, Scotland, France, Denmark, western Germany and part of Switzerland.

The first map always duplicates rainviewer very accurately. They second map is great if displaying areas of high rainfall (green, yellow and red on PiClock) and they look accurate. Lower levels that the Rainviewer on my browser colour green are pale to dark blue and much larger on PiClock. They look like "blobs" rather than the well defined images on Rainviewer.

I have updated to the latest FW with git pull.

SerBrynden, 2023-06-26:
RainViewer will look different in the web browser or on the mobile app.  The RainViewer API doesn't allow one to tweak the minimal precipitation intensity.  Also, the color scheme might be different.  You can change the radar color scheme in the PiClock config file and the schemes can be found here: https://www.rainviewer.com/api/color-schemes.html

SerBrynden, 2023-07-18:
I haven't really tried the other color schemes, but it's easy enough to change it in the config file and find one you like.

closed 2023-07-18 by SerBrynden

feh123, 2023-07-18:
Yes - thanks for the info. I picked the original and although less colourful it makes the depiction of the rainfall on the two radar maps seem more aligned. I guess this maybe just my feeling!