SerBrynden/PiClock #6: Rainviewer Radar Error

feh123, 2023-07-20:
This morning my PiClock was not displaying rain radar and I do not see the rainviewer box at the top of my two radar maps. I checked the startup text and found:

radar1... get radar tiles for time 1689836400 (2023-07-20 08:00:00+01:00)
radar1 tile0 https://tilecache.rainviewer.com/v2/radar/1689836400//256/7/63/41/6/1_1.png
ERROR from rainviewer.com: <?xml version="1.0" encoding="iso-8859-1"?>
<!DOCTYPE html PUBLIC "-//W3C//DTD XHTML 1.0 Transitional//EN"
"http://www.w3.org/TR/xhtml1/DTD/xhtml1-transitional.dtd">
<html xmlns="http://www.w3.org/1999/xhtml" xml:lang="en" lang="en">
<head>
<title>404 - Map not found</title>
Radar2 shows the same error. Other than the radar the PiClock screen is showing all the usual data. Rainviewer does show the live map in my laptop browser so I don't think I have any connection issues at my end. Any thoughts on what is happening? Thanks.

feh123, 2023-07-20:
Now two hours later the radar rainful pattern and the rainviewer.com logo have returned. Clearly some sort of issue at rainviewer or in the connection path. Glad to have the radar back!

closed 2023-07-20 by feh123

SerBrynden, 2023-07-20:
Ok.  Just saying it might have resolved sooner if you restarted PiClock or rebooted the whole device right away.