SerBrynden/PiClock #7: Add 2 line codes to PyQtPiClock.py

parallelbgls, 2023-07-28:
from line 1566 in PyQtPiClock.py

```python
def get_local(self, path):
        try:
            dir_content = os.listdir(path)
            for each in dir_content:
                full_file = os.path.join(path, each)
                if os.path.isfile(full_file) and (full_file.lower().endswith('png')
                                                  or full_file.lower().endswith('jpg')):
                    self.img_list.append(full_file)
                elif os.path.isdir(full_file):  # Recursively get images inside folder
                    self.get_local(full_file)
        except OSError:
            print(traceback.format_exc())
```

Sorry I don't have time to folk this project and pull request those codes.

SerBrynden, 2023-07-28:
@parallelbgls What is the purpose of this? What problem does it fix?

parallelbgls, 2023-07-28:
My photo booth folder is like:
/home/rpi/Pictures/Slides/2022/2022-XXX/1.jpg
/home/rpi/Pictures/Slides/2022/2022-XXX/2.jpg
/home/rpi/Pictures/Slides/2022/2022-YYY/1.jpg
/home/rpi/Pictures/Slides/2023/2023-XXX/1.jpg
/home/rpi/Pictures/Slides/2023/2023-YYY/1.jpg

So no pictures will be shown without these appending codes if I wrote /home/rpi/Pictures/Slides in Config.py,
because there is no jpg file in /home/rpi/Pictures/Slides but only folders.

parallelbgls, 2023-07-28:
By the way folders inside Slides are cids mount from synology nas.

parallelbgls, 2023-09-08:
Sorry, I don't think this is a good suggestion so I will close this issue.

closed 2023-09-08 by parallelbgls