SerBrynden/PiClock #8: Adding a second marker to the Radar

xxxDrewedxxx, 2023-08-25:
Is there a way to display 2 markers on the same radar?  Say one for home and one for a cabin in the woods, or whatever?

SerBrynden, 2023-08-25:
I think you can.  In the config file under each radar section, try experimenting with adding another set of marker parameters.  Like this: 
```
radar1 = {
    'center': radar_location,  # the center of your radar block
    'zoom': 10,  # this is a maps zoom factor, bigger = smaller area
    'basemap': map_base,  # Mapbox style for standard map or custom map with land and water only
    'overlay': map_overlay,  # Mapbox style for labels, roads, and borders only
    'color': 6,  # rainviewer radar color style:
    # https://www.rainviewer.com/api.html#colorSchemes
    'smooth': 1,  # rainviewer radar smoothing
    'snow': 1,  # rainviewer radar show snow as different color
    'markers': (  # google maps markers can be overlaid
        {
            'visible': 1,
            'location': radar_location,
            'color': 'red',
            'size': 'small',
            'image': 'teardrop-dot',
        }, 
        {
            'visible': 1, 
            'location': cabin_location,
            'color': 'blue',
            'size': 'small',
            'image': 'teardrop-dot',
        },  # dangling comma is on purpose.
    )
}
```
The `location` must be a LatLng object.  See how it's done for `radar_location = LatLng(primary_coordinates[0], primary_coordinates[1])` at the top of config.

xxxDrewedxxx, 2023-08-25:
Yep!  That did it.  I could swear that I tried that before, but I probably goofed up the spacing and gave it up as not working.  lol
Thanks!

closed 2023-08-25 by SerBrynden