SerBrynden/PiClock #9: PiClock crashing to desktop, on occasion, due to METAR (?)

Vindictor, 2023-11-28:
Hi there,
In the previous thread, on issues with N0bel's PiClock, after I got my "SerBrynden PiClock" working on Bookworm on a Pi4, I mentioned that I'd had a crash to the desktop, which looked like it was due to METAR. 

Over the long period of time I used the N0bel PiClock I'd sometimes have a small issue where the METAR data would be missing, or maybe have one or two elements of data missing.
Later on it would all be working again. I just ignored it. 

I'm just guessing, but it seems like when this issue occurs with your fork, instead of the data simply not displaying, and me seeing a blank area in the top left of my screen, it completely crashes to the desktop. 

This doesn't generally happen often. It happened while we were talking on that previous thread a couple of weeks ago. 
It happened once in the interim, until yesterday.  I had a crash to desktop yesterday evening, and it's crashing to the desktop right now. 

Again, and to be clear, it the PiClock does work most of the time. It was on all day yesterday, until it crashed after 7pm. 
It was on all day today, functioning normally, until approx 5:20pm, just now. 

Yesterday after I'd noticed the PiClock screen was on the desktop I restarted it, and it started working. 
Today it will not.  This is what happened to me before. At some point the METAR data will be OK, and PiClock will launch. 
At the moment it starts to launch, I see the background image and clock appear, and a spit second later it crashes to desktop again. 
If I comment out METAR in my config file, the PiClock will start again.   or as I said, if I just wait until whatever is causing the issue passes, then PiClock will launch again. 
Obviously the missing, malformed, or whatever is wrong with the METAR data cannot be fixed... however, displaying nothing in the corner of the screen is preferable to an all out crash to the desktop.  

I had a look at the log for yesterday afternoon.  Full of the usual stuff, weather updates.... but then at 7:12pm I got this (I will edited out identifying information, however, if you do want my METAR code for testing, maybe I can send it to you.  

Just for reference, now that I come to look more closely at the log output below, I can see the metar .txt file URL.
I just copied this URL and pasted it into a browser.  It came back with a result.  I  just re-launched PiClock and it works again. 
This is at 6:41pm, so approx 20 mins after the first crash..... OK, back to the log....

So, from yesterday evening:
[2023-11-27 19:12:04.481445+00:00] getting METAR current conditions: https://tgftp.nws.noaa.gov/data/observations/metar/stations/xxx.TXT
[2023-11-27 19:12:04.486791+00:00] getting OpenWeather forecast: https://api.openweathermap.org/data/2.5/forecast?appid=xxx&lat=xxx&lon=-xxx&units=imperial&lang=en&r=0.8356474405966696
[2023-11-27 19:12:05.220796+00:00] wxmetar: xxx 271850Z 01011KT 8000 SCT022 BKN042 09/06 Q1007 TEMPO 4000 -SHRA SCT01200 BKN2000 RMK WHT TEMPO GRN
[2023-11-27 19:12:05.231953+00:00] Traceback (most recent call last):
[2023-11-27 19:12:05.236675+00:00] File "/home/xxx/PiClock/Clock/PyQtPiClock.py", line 1229, in wxfinished_metar
[2023-11-27 19:12:05.241463+00:00] f = Metar.Metar(wxstr)
[2023-11-27 19:12:05.246158+00:00] ^^^^^^^^^^^^^^^^^^
[2023-11-27 19:12:05.251594+00:00] File "/home/xxx/PiClock/venv/lib/python3.11/site-packages/metar/Metar.py", line 417, in __init__
[2023-11-27 19:12:05.256634+00:00] raise ParserError("Unparsed groups in body: "+code)
[2023-11-27 19:12:05.263294+00:00] metar.Metar.ParserError: Unparsed groups in body: SCT01200 BKN2000

so that was yesterday evening.  One crash, it started fine after this, and had been running for days, or even a week without issue.

Then, this evening, 
crash 1:

[2023-11-28 17:15:25.705642+00:00] getting METAR current conditions: https://tgftp.nws.noaa.gov/data/observations/metar/stations/xxx.TXT
[2023-11-28 17:15:25.716788+00:00] getting OpenWeather forecast: https://api.openweathermap.org/data/2.5/forecast?appid=xxx&lat=xxx&lon=-xxx&units=imperial&lang=en&r=0.7948366255739239
[2023-11-28 17:15:26.785366+00:00] wxmetar: xxx 281650Z 23001KT 9999 VCSH FEW012 SCT040 BKN060 07/05 Q1010 6000 -SHRA FEW010 SCT018TCU
[2023-11-28 17:15:26.795680+00:00] Traceback (most recent call last):
[2023-11-28 17:15:26.800158+00:00] File "/home/xxx/PiClock/Clock/PyQtPiClock.py", line 1229, in wxfinished_metar
[2023-11-28 17:15:26.804436+00:00] f = Metar.Metar(wxstr)
[2023-11-28 17:15:26.808839+00:00] ^^^^^^^^^^^^^^^^^^
[2023-11-28 17:15:26.814022+00:00] File "/home/xxx/PiClock/venv/lib/python3.11/site-packages/metar/Metar.py", line 417, in __init__
[2023-11-28 17:15:26.819076+00:00] raise ParserError("Unparsed groups in body: "+code)
[2023-11-28 17:15:26.824128+00:00] metar.Metar.ParserError: Unparsed groups in body: -SHRA FEW010 SCT018TCU

Just below 10 min later I tried to start it again:

[2023-11-28 17:24:54.445368+00:00] wxmetar: xxx 281650Z 23001KT 9999 VCSH FEW012 SCT040 BKN060 07/05 Q1010 6000 -SHRA FEW010 SCT018TCU
[2023-11-28 17:24:54.450040+00:00] Traceback (most recent call last):
[2023-11-28 17:24:54.454619+00:00] File "/home/xxx/PiClock/Clock/PyQtPiClock.py", line 1229, in wxfinished_metar
[2023-11-28 17:24:54.459088+00:00] f = Metar.Metar(wxstr)
[2023-11-28 17:24:54.464160+00:00] ^^^^^^^^^^^^^^^^^^
[2023-11-28 17:24:54.469015+00:00] File "/home/xxx/PiClock/venv/lib/python3.11/site-packages/metar/Metar.py", line 417, in __init__
[2023-11-28 17:24:54.473833+00:00] raise ParserError("Unparsed groups in body: "+code)
[2023-11-28 17:24:54.479259+00:00] metar.Metar.ParserError: Unparsed groups in body: -SHRA FEW010 SCT018TCU

I relaunched PiClock again at approx 5:40pm, and it's working fine, with all METAR data displayed as usual. 

Hopefully these Unparsed groups in body mean something to you. 
My uneducated guess is that it's receiving data it doesn't know what to do with, and rather than display nothing, it's crashing to the desktop. 

My huge appreciation if you can have a look at this.  
Not sure if I can send a private message via Github, otherwise I guess it won't hurt too much to share the metar code (although again, it works fine, mostly)

OK, just before I submit this post, I've gone back to the latest log.  I mentioned that it's running perfectly again, now. 
I've grabbed the metar part of the log, to show how it looks when there isn't an issue.  After this the log goes on to radar tiles. 

[2023-11-28 17:40:20.138503+00:00] getting METAR current conditions: https://tgftp.nws.noaa.gov/data/observations/metar/stations/xxx.TXT
[2023-11-28 17:40:20.184237+00:00] getting OpenWeather One Call: https://api.openweathermap.org/data/2.5/onecall?appid=xxx&lat=xxx&lon=-xxx&units=imperial&lang=en&r=0.5414945816543562
[2023-11-28 17:40:20.194492+00:00] wxstart for radar1
[2023-11-28 17:40:20.199289+00:00] wxstart for radar2
[2023-11-28 17:40:20.318915+00:00] ERROR from api.openweathermap.org: 401 - Invalid API key. Please see https://openweathermap.org/faq#error401 for more info.
[2023-11-28 17:40:20.323493+00:00] OpenWeather One Call failed... switching to Current Weather and Forecast
[2023-11-28 17:40:20.328397+00:00] getting OpenWeather forecast: https://api.openweathermap.org/data/2.5/forecast?appid=xxx&lat=xxx&lon=-xxx&units=imperial&lang=en&r=0.31917036963541157
[2023-11-28 17:40:21.050865+00:00] wxmetar: xxx 281720Z 33002KT 9999 VCSH FEW012 FEW018TCU SCT020 BKN060 08/06 Q1010 NOSIG RMK WHT WHT

(I always see that invalid API key error for OpenWeather, however, I always get the Openweather forecast.  It doesn't appear to be affecting this Metar issue, anyway) (Of course I've edited out the API key before posting)

Thanks again.

SerBrynden, 2023-11-28:
Moving straight to your error log outputs:

The problem is with the METAR Python package, not PiClock.  Some METAR stations add extra items to the code string.  The METAR package raises an error when unparsable groups are found in the METAR code string or an unexpected exception is encountered.  I've added the option of `strict=False` to the METAR object instance in PiClock, which will prevent exceptions from being raised and only generate warning messages.  To update, do 
```
cd PiClock
git pull
bash update.sh
```

The Invalid API key message can be ignored and just means your key is not compatible with the legacy OWM 2.5 OneCall (which has both current weather conditions and forecast) so it falls back to using the separate OWM calls, one for current weather and one for forecast.  However, since you're also using METAR, the OWM current weather call is skipped in favor for the METAR current conditions.

SerBrynden, 2023-12-01:
Your METAR station is wonky.  It reports the atmospheric pressure as `Q////` which is jibberish.  Now we have to include error handling just for METAR pressure and skip it if it can't be read.  

I also added more error handling for index errors with the radar tiles.  The index errors will still be reported in the log, but the program should continue on.

In your log, there were two index errors.  The first one was handled with a try/except clause and the program continued, but the second one hadn't been noticed before.

Do a `git pull` to update.

Vindictor, 2023-12-01:
OK, great.  I'll log in and update that, shortly. 
I think I've mentioned before that I've had METAR issues for.... as long as I've been using it. 
In the n0bel version it would just leave areas of the current forecast blank.  Sometimes I'd have no current forecast data, and from memory, sometimes I'd get the temperature, and maybe the rest was blank.  I just ignored it, as I knew it would be fine again after it updated again.   I didn't mind as the PiClock kept on running.  it's the crashing to the desktop which is the biggest issue.  Maybe this is related to it now running in a virtual environment, compared to my old PiClock which was still running on Bulleye, or older. 
As for my local METAR station... it's a Royal Navy base. Maybe the Royal Navy can't afford Pi's yet, so they're running their METAR system from a C64.   It's a weird issue, usually it functions perfectly.  A week can go by without a glitch, and then I can have two updates in a day which will cause an issue. 
As I type this, it's been running fine since I restarted it this morning, and while typing this message the METAR info has just updated again. 

Regardless, I'll do a git pull and update in a short while. 

Thank you, very much.

closed 2023-12-14 by SerBrynden

Vindictor, 2024-01-05:
Unfortunately I seem to be running into a similar issue, again (?)

A day or two ago I noticed that my PiClock had randomly closed, and the screen was sitting on the PiOS desktop.

I SSH'd in and performed a git pull, and a bash update.sh

There was an update, so I thought perhaps this had fixed the issue.
I ran PiClock again, and all seemed well.

It was working fine today as well, until a short while ago.
I glanced up to see the screen sitting at the PiOS desktop, again.

I SSH'd in again, did another git pull, there seemed to be another update.  Another bash update.sh.
But now when I run PiClock, it loads briefly (as with the previous issue, above) the Clock can be seen, and the background picture, and the Openweathermap forecast.... but it looks like when it loads the METAR info, it crashes to the desktop, again.

I just tried to run it twice.

All that's in the log, is this

2024-01-05 15:35:37.133530 GMT (UTC+0000) - INFO: map base url for radar1: http://maps.googleapis.com/maps/api/staticmap?key=xxxx&center=xxxx,-xxxx&zoom=7&size=400x330&maptype=hybrid
2024-01-05 15:35:37.137696 GMT (UTC+0000) - INFO: map base url for radar2: http://maps.googleapis.com/maps/api/staticmap?key=xxxx&center=xxxx,-xxxx&zoom=11&size=400x330&maptype=hybrid
2024-01-05 15:35:37.141770 GMT (UTC+0000) - INFO: map base url for radar3: http://maps.googleapis.com/maps/api/staticmap?key=xxxx&center=xxxx,-xxxx&zoom=6&size=466x420&maptype=hybrid
2024-01-05 15:35:37.145759 GMT (UTC+0000) - INFO: map base url for radar4: http://maps.googleapis.com/maps/api/staticmap?key=xxxx&center=xxxx,-xxxx&zoom=10&size=466x420&maptype=hybrid
2024-01-05 15:35:37.783517 GMT (UTC+0000) - libpng warning: iCCP: known incorrect sRGB profile
2024-01-05 15:35:40.413800 GMT (UTC+0000) - INFO: getting METAR current conditions: https://tgftp.nws.noaa.gov/data/observations/metar/stations/xxxx.TXT
2024-01-05 15:35:40.467852 GMT (UTC+0000) - INFO: getting OpenWeather One Call: https://api.openweathermap.org/data/2.5/onecall?appid=xxxx&lat=xxxx&lon=-xxxx&units=imperial&lang=en&r=0.5901926187758628
2024-01-05 15:35:40.474519 GMT (UTC+0000) - INFO: wxstart for radar1
2024-01-05 15:35:40.479876 GMT (UTC+0000) - INFO: wxstart for radar2
2024-01-05 15:35:40.685697 GMT (UTC+0000) - WARNING: Response from api.openweathermap.org: 401 - Invalid API key. Please see https://openweathermap.org/faq#error401 for more info.
2024-01-05 15:35:40.690955 GMT (UTC+0000) - WARNING: OpenWeather One Call failed...
2024-01-05 15:35:40.695941 GMT (UTC+0000) - WARNING: Falling back to separate OpenWeather calls for current weather conditions and forecast
2024-01-05 15:35:40.700812 GMT (UTC+0000) - INFO: getting OpenWeather forecast: https://api.openweathermap.org/data/2.5/forecast?xxxx&lat=xxxx&lon=-xxxx&units=imperial&lang=en&r=0.20717837289492247
2024-01-05 15:35:41.037885 GMT (UTC+0000) - INFO: wxmetar: xxxx 051520Z AUTO /////KT 9999 FEW047/// 09/04 Q1008
2024-01-05 15:35:41.048743 GMT (UTC+0000) - Traceback (most recent call last):
2024-01-05 15:35:41.053612 GMT (UTC+0000) - File "/home/xxx/PiClock/Clock/PyQtPiClock.py", line 1305, in wxfinished_metar
2024-01-05 15:35:41.058505 GMT (UTC+0000) - ws = (Config.LWind + wd + ' ' + '%.1f' % (f.wind_speed.value('KMH')) + 'km/h')
2024-01-05 15:35:41.063354 GMT (UTC+0000) - ^^^^^^^^^^^^^^^^^^
2024-01-05 15:35:41.068566 GMT (UTC+0000) - AttributeError: 'NoneType' object has no attribute 'value'

and the second attempt....

2024-01-05 15:39:27.704910 GMT (UTC+0000) - INFO: map base url for radar1: http://maps.googleapis.com/maps/api/staticmap?key=xxxx&center=xxxx,-xxxx&zoom=7&size=400x330&maptype=hybrid
2024-01-05 15:39:27.709065 GMT (UTC+0000) - INFO: map base url for radar2: http://maps.googleapis.com/maps/api/staticmap?key=xxxx&center=xxxx,-xxxx&zoom=11&size=400x330&maptype=hybrid
2024-01-05 15:39:27.713351 GMT (UTC+0000) - INFO: map base url for radar3: http://maps.googleapis.com/maps/api/staticmap?key=xxxx&center=xxxx,-xxxx&zoom=6&size=466x420&maptype=hybrid
2024-01-05 15:39:27.717447 GMT (UTC+0000) - INFO: map base url for radar4: http://maps.googleapis.com/maps/api/staticmap?key=xxxx&center=xxxx,-xxxx&zoom=10&size=466x420&maptype=hybrid
2024-01-05 15:39:28.327080 GMT (UTC+0000) - libpng warning: iCCP: known incorrect sRGB profile
2024-01-05 15:39:30.924637 GMT (UTC+0000) - INFO: getting METAR current conditions: https://tgftp.nws.noaa.gov/data/observations/metar/stations/xxxx.TXT
2024-01-05 15:39:30.973125 GMT (UTC+0000) - INFO: getting OpenWeather One Call: https://api.openweathermap.org/data/2.5/onecall?appid=xxxx&lat=xxxx&lon=-xxxx&units=imperial&lang=en&r=0.6063000165373416
2024-01-05 15:39:30.978680 GMT (UTC+0000) - INFO: wxstart for radar1
2024-01-05 15:39:30.984801 GMT (UTC+0000) - INFO: wxstart for radar2
2024-01-05 15:39:31.141869 GMT (UTC+0000) - WARNING: Response from api.openweathermap.org: 401 - Invalid API key. Please see https://openweathermap.org/faq#error401 for more info.
2024-01-05 15:39:31.145967 GMT (UTC+0000) - WARNING: OpenWeather One Call failed...
2024-01-05 15:39:31.150447 GMT (UTC+0000) - WARNING: Falling back to separate OpenWeather calls for current weather conditions and forecast
2024-01-05 15:39:31.154228 GMT (UTC+0000) - INFO: getting OpenWeather forecast: https://api.openweathermap.org/data/2.5/forecast?appid=xxxx&lat=xxxx&lon=-xxxx&units=imperial&lang=en&r=0.3651581171587942
2024-01-05 15:39:32.281712 GMT (UTC+0000) - INFO: wxmetar: xxxx 051520Z AUTO /////KT 9999 FEW047/// 09/04 Q1008
[... cut ...]
2024-01-05 15:39:32.300933 GMT (UTC+0000) - ^^^^^^^^^^^^^^^^^^
2024-01-05 15:39:32.305714 GMT (UTC+0000) - AttributeError: 'NoneType' object has no attribute 'value'

I hope this can help narrow down the issue.
Up until yesterday, or possibly the day before. it's been working perfectly ever since you added the fix.
But now I have the crashing, again.

Presumably if I wait, as before, then when the METAR data changes I should be able to launch PiClock, again.

My apologies for the continuing issues,
With thanks, and best regards.