SerBrynden/PiClock #10: modules not loading

youkiddin, 2023-12-10:
hi i have pi4 that's been updated to bookworm and trying to get this to see my external ds18b20 temp sensors. the temp driver is installed and the boot/config.txt has been edited along with /etc/modules.on start up i get this. any idea where to look?                                                                     bookworm@raspberrypi:~ $ cd PiClock
bookworm@raspberrypi:~/PiClock $ bash startup.sh -n -s
Disabling screen blanking...
server does not have extension for -dpms option
Setting sound to max (assuming Monitor Tv controls volume)...
Activating virtual environment...
Checking for NeoPixels Ambilight...
Checking for GPIO Buttons...
Checking for Temperature Sensors...
Starting Temperature Service...
Starting PiClock... logging to screen.
Traceback (most recent call last):
  File "/home/bookworm/PiClock/Temperature/TempServer.py", line 11, in <module>
    from BaseHTTPServer import BaseHTTPRequestHandler, HTTPServer
ModuleNotFoundError: No module named 'BaseHTTPServer'
Traceback (most recent call last):
  File "/home/bookworm/PiClock/Clock/PyQtPiClock.py", line 30, in <module>
    import ApiKeys  # NOQA
    ^^^^^^^^^^^^^^
ModuleNotFoundError: No module named 'ApiKeys'
bookworm@raspberrypi:~/PiClock $

SerBrynden, 2023-12-11:
This http server package wasn't changed when PiClock was updated from Python2 to Python3.  
It's been fixed.  
Do a `git pull` to update.

youkiddin, 2023-12-11:
Ok thanks I will do that

closed 2023-12-12 by youkiddin