SerBrynden/PiClock #11: not able to launch PiClock

jamesdoesstuff2008, 2024-01-02:
hey, i am unable to launch PiClock. could you see what the problem is?
`carolinecheney@raspberrypi:~/PiClock $ bash startup.sh -n -s
Disabling screen blanking...
Setting sound to max (assuming Monitor Tv controls volume)...
Activating virtual environment...
Checking for NeoPixels Ambilight...
1795
1797
Checking for GPIO Buttons...
Starting gpio-keys Service...
Checking for Temperature Sensors...
ioctl UI_DEV_CREATE: Invalid argument
Starting PiClock... logging to screen.
ERROR: Config file not found Config.py`

SerBrynden, 2024-01-02:
Did you copy the example config file and edit it with your own settings?  See here: https://github.com/SerBrynden/PiClock/blob/master/Documentation/Install.md#configure-your-piclock

Also, make sure you did the same with the API key file: https://github.com/SerBrynden/PiClock/blob/master/Documentation/Install.md#configure-the-piclock-api-keys

jamesdoesstuff2008, 2024-01-02:
thanks, i forgot to copy the files over and that was the issue. thanks a
bunch!

jamesdoesstuff2008, 2024-01-02:
but i do have another problem, when i press f2 the radio stream dose not
work.

SerBrynden, 2024-01-02:
What link did you put in the config file for the audio stream?  Did you install the mpg123 Audio Streamer per the guide: https://github.com/SerBrynden/PiClock/blob/master/Documentation/Install.md#optional---mpg123-audio-streaming ?

jamesdoesstuff2008, 2024-01-02:
> Documentation/Install.md#optional---mpg123-audio-streaming ?

Never mind I was able to fix it.  But I am also having a hard time figuring
out how to launch the software on boot and changing what keys do what.

SerBrynden, 2024-01-02:
What part of the install guide are you hung up on?

jamesdoesstuff2008, 2024-01-02:
You got a social I could message you in so this feed won't be too long?

https://ci5.googleusercontent.com/proxy/qg0bnoNP0Nnn5DWx9lUMBMtCcsMUWLzqeSP3IwpftE2PhUi5xgk47aMV1L3UgMO9NmMpxYl__h5xxkq_VywvLxnwoS0Xvd2u-W0BN9V78RTwdfri02guy0lO61zmCWACaZu2jXtFAVd2QEKsXvY7lMQcPucvoj6WOGC8nlN4u2BmYB91Y2XCGmU7PDsWPSyhpJiDc1vvi5zQVfJI5qhHaz1WQTdyfSXP_kVmh1vcNxJD1BwMrw=s0-d-e1-ft#https://github.com/notifications/beacon/BEXC3MPVVFUBRXXKAP5YTWTYMR7IHA5CNFSM6AAAAABBJVGNGWWGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTTPXRO42.gif>Message
ID: ***@***.***>

SerBrynden, 2024-01-02:
Best to track everything here.

jamesdoesstuff2008, 2024-01-02:
Ok then.

https://ci5.googleusercontent.com/proxy/zSX_Irka70KPiumCNkgOWUC76-Gm496I7QzHk792HnEmlZGXxERV4smAe4O_iYGOgYkSmjuDpgHNNnDe4ro9YIeC5_-wpRbq5yOhppb7gEWDZ3Rq0SJZyE9N7y9_43cOUpC-5VT9VhOGyQ5VlgIQytngaUoQD6wiLRneyZENCiJ5Pe0GRS6HdQ769rGe1WzHBThS4Nm1UsAgUHELUQENdppDAg-spa2EYBwyjh3f4grtwcWvnA=s0-d-e1-ft#https://github.com/notifications/beacon/BEXC3MMIRVNGYQUA52Z4OOLYMSIGXA5CNFSM6AAAAABBJVGNGWWGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTTPXVAOS.gif>Message
ID: ***@***.***>

jamesdoesstuff2008, 2024-01-02:
I followed the instructions on open on boot but it doesn't work and have to
open it manually via cmd.

https://ci5.googleusercontent.com/proxy/qg0bnoNP0Nnn5DWx9lUMBMtCcsMUWLzqeSP3IwpftE2PhUi5xgk47aMV1L3UgMO9NmMpxYl__h5xxkq_VywvLxnwoS0Xvd2u-W0BN9V78RTwdfri02guy0lO61zmCWACaZu2jXtFAVd2QEKsXvY7lMQcPucvoj6WOGC8nlN4u2BmYB91Y2XCGmU7PDsWPSyhpJiDc1vvi5zQVfJI5qhHaz1WQTdyfSXP_kVmh1vcNxJD1BwMrw=s0-d-e1-ft#https://github.com/notifications/beacon/BEXC3MPVVFUBRXXKAP5YTWTYMR7IHA5CNFSM6AAAAABBJVGNGWWGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTTPXRO42.gif>Message
ID: ***@***.***>

SerBrynden, 2024-01-02:
So, when you open a new terminal and input the commands found under [Autostart Method 1](https://github.com/SerBrynden/PiClock/blob/master/Documentation/Install.md#autostart-method-1), do you get any errors?  Do you get a new PiClock icon on the desktop?  You can just copy the whole block of commands and paste them into the terminal.

jamesdoesstuff2008, 2024-01-02:
Ill try that and get back to you

> So, when you open a new terminal and input the commands found under
Autostart Method 1, do you get any errors? Do you get a new PiClock icon on
the desktop? You can just copy the whole block of commands and paste them
into the terminal.
https://ci6.googleusercontent.com/proxy/C1GtV6IUDfkmDBG-Maj0ot_hBdDLlnAGOao5SuqdTbTHXwxIDPPKsIQ92ZlewESJHLVgJmi2ImsLPyWiyJX9D2kaFaLGX5xeB4OIP9PbeBKlI69zNPW1FZcnis_y_-0U2WriEgpeJHXwdumaAAi_Gn2fVBfX7_RU0l89LvDhC0z5yAImD2lED7NEGuEztJd8ix0b6yxucNzsBFSoruZgFV9wLeDi4YWkGPtdOxtb9OT_donDAQ=s0-d-e1-ft#https://github.com/notifications/beacon/BEXC3MJKD5OV57H2I6F3P5DYMSLHHA5CNFSM6AAAAABBJVGNGWWGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTTPXV7S4.gif>Message
ID: ***@***.***>

jamesdoesstuff2008, 2024-01-02:
So… now what?

maserowik, 2024-01-03:
For me to get auto boot to work I had to use the other method of adding the @reboot commands to my Crontab 

Sent from the all new AOL app for iOS

jamesdoesstuff2008, 2024-01-03:
> BEXC3MPJZZ334YYVNPTPL23YMSLHHAVCNFSM6AAAAABBJVGNGWVHI2DSMVQW

I’m new to programming, can you tell me how you do that?

maserowik, 2024-01-03:
Autostart Method 2

To have it auto start on boot you need to do one more thing, edit the crontab file as follows: (it will automatically start nano) (NOT as root)
crontab -e

and add the following line:
@reboot bash /home/pi/PiClock/startup.sh
press ctrl + x to savesudo reboot

jamesdoesstuff2008, 2024-01-03:
> BEXC3MNBB6HE2GCQ2USWQGLYMSNYJAVCNFSM6AAAAABBJVGNGWVHI2DSMVQW

Do you have to change pi to your pi’s name?

maserowik, 2024-01-03:
Not following

 

I named my Pi Zero to Pi-Clock

 

Using the default user name of Pi

 

This email has been checked for viruses by Avast antivirus software.
www.avast.com

jamesdoesstuff2008, 2024-01-03:
> BEXC3MNADMRSGCB5OIDFCWLYMSPANAVCNFSM6AAAAABBJVGNGWVHI2DSMVQW
> AGEFY5W7VKPBFCRSD5KCVZLYMSP2RAVCNFSM6AAAAABBJVGNGWVHI2DSMVQW
> IX3LMV43OSLTON2WKQ3PNVWWK3TUHMYTQNZUG4YTINZRG4> .
> notifications/beacon/AGEFY5UWFISJAP522FEGH73YMSP2RA
> 5CNFSM6AAAAABBJVGNGWWGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23LFNZ

Ok, so after adding that I just restart it?

maserowik, 2024-01-03:
Correct issue 

Sudo reboot once you same the crontab

jamesdoesstuff2008, 2024-01-03:
> SerBrynden/PiClock/blob/%20%3chttps:/github.com/%20%0b> <
> BEXC3MJ5P5H5COPIHMNA643YMSQURAVCNFSM6AAAAABBJVGNGWVHI2DSMVQW
> AGEFY5WE5RWXMBXVJSV6CZ3YMSQ53AVCNFSM6AAAAABBJVGNGWVHI2DSMVQW
> IX3LMV43OSLTON2WKQ3PNVWWK3TUHMYTQNZUG4ZDAOJRGA> .
> notifications/beacon/AGEFY5RASK32GQ6DEII222TYMSQ53A

How long does it take to boot? Waited about a minute.

maserowik, 2024-01-03:
Did you also follow these steps

The ones  in bold are a must.

 

 

 

First boot and configure

A keyboard and mouse are really handy at this point. When you first boot your Pi, you'll be presented with the desktop. Following this there will be several prompts to set things up, follow those prompts and set things as they make sense for you. Of course setting the proper timezone for a clock is key.

 

Eventually the Pi will reboot, and you'll be back to the desktop. You need to configure a few more things.

 

Navigate to Menu->Preferences->Raspberry Pi Configuration. Just change the Items below.

 

System Tab

Hostname: (Maybe set this to PiClock?)

Boot: To Desktop

Auto Login: Checked

Overscan: (Initially leave as default, but if your monitor has extra black area on the border, or bleeds off the edge, then change this)

Interfaces

1-Wire Enable (for the inside temperature, DS18B20 if you're using it)

SSH is handy (if you'd like to connect to your clock from another computer)

VNC can be handy (same reason as ssh)

Click ok, and allow it to reboot.

jamesdoesstuff2008, 2024-01-03:
I opened the PiClock.desktop and ran it and got a error code 126 with
permission denied.

SerBrynden, 2024-01-03:
Anyplace you see `/home/pi/PiClock/startup.sh`, replace `pi` with whatever username you are using. It has to match the directory where PiClock is saved.

jamesdoesstuff2008, 2024-01-03:
Imma just make things easy and rename to pi

jamesdoesstuff2008, 2024-01-03:
Ok, fixed that, now I can not figure out how to use the IR sensor.  Could I
have a little more in depth tutorial?

SerBrynden, 2024-01-03:
You're better off using a small keyboard or having a small inexpensive touchscreen monitor for Raspberry Pi in order to change the screens.