SerBrynden/PiClock #12: PiClock crashing to desktop, on occasion (again), due to METAR (?)

Vindictor, 2024-01-05:
Hi there,
Sorry, I initally posted this message in the thread I'd previously created about having PiClock crash to desktop due to METAR.
Not being well versed with GitHub I'd thought that perhaps replying to that thread again would reopen the issue.
Obviously this did not happen, and so I'm unsure whether @SerBrynden will get notified of the new reply, so I'm opening this new issue.  My apologies if this is considered impolite.

This is in continuation of
[https://github.com/SerBrynden/PiClock/issues/9](url)

Unfortunately I seem to be running into a similar issue, again (?)

A day or two ago I noticed that my PiClock had randomly closed, and the screen was sitting on the PiOS desktop.

I SSH'd in and performed a git pull, and a bash update.sh

There was an update, so I thought perhaps this had fixed the issue.
I ran PiClock again, and all seemed well.

It was working fine today as well, until a short while ago.
I glanced up to see the screen sitting at the PiOS desktop, again.

I SSH'd in again, did another git pull, there seemed to be another update. Another bash update.sh.
But now when I run PiClock, it loads briefly (as with the previous issue, above) the Clock can be seen, and the background picture, and the Openweathermap forecast.... but it looks like when it loads the METAR info, it crashes to the desktop, again.

I just tried to run it twice.

All that's in the log, is this

2024-01-05 15:35:37.133530 GMT (UTC+0000) - INFO: map base url for radar1: http://maps.googleapis.com/maps/api/staticmap?key=xxxx&center=xxxx,-xxxx&zoom=7&size=400x330&maptype=hybrid
2024-01-05 15:35:37.137696 GMT (UTC+0000) - INFO: map base url for radar2: http://maps.googleapis.com/maps/api/staticmap?key=xxxx&center=xxxx,-xxxx&zoom=11&size=400x330&maptype=hybrid
2024-01-05 15:35:37.141770 GMT (UTC+0000) - INFO: map base url for radar3: http://maps.googleapis.com/maps/api/staticmap?key=xxxx&center=xxxx,-xxxx&zoom=6&size=466x420&maptype=hybrid
2024-01-05 15:35:37.145759 GMT (UTC+0000) - INFO: map base url for radar4: http://maps.googleapis.com/maps/api/staticmap?key=xxxx&center=xxxx,-xxxx&zoom=10&size=466x420&maptype=hybrid
2024-01-05 15:35:37.783517 GMT (UTC+0000) - libpng warning: iCCP: known incorrect sRGB profile
2024-01-05 15:35:40.413800 GMT (UTC+0000) - INFO: getting METAR current conditions: https://tgftp.nws.noaa.gov/data/observations/metar/stations/xxxx.TXT
2024-01-05 15:35:40.467852 GMT (UTC+0000) - INFO: getting OpenWeather One Call: https://api.openweathermap.org/data/2.5/onecall?appid=xxxx&lat=xxxx&lon=-xxxx&units=imperial&lang=en&r=0.5901926187758628
2024-01-05 15:35:40.474519 GMT (UTC+0000) - INFO: wxstart for radar1
2024-01-05 15:35:40.479876 GMT (UTC+0000) - INFO: wxstart for radar2
2024-01-05 15:35:40.685697 GMT (UTC+0000) - WARNING: Response from api.openweathermap.org: 401 - Invalid API key. Please see https://openweathermap.org/faq#error401 for more info.
2024-01-05 15:35:40.690955 GMT (UTC+0000) - WARNING: OpenWeather One Call failed...
2024-01-05 15:35:40.695941 GMT (UTC+0000) - WARNING: Falling back to separate OpenWeather calls for current weather conditions and forecast
2024-01-05 15:35:40.700812 GMT (UTC+0000) - INFO: getting OpenWeather forecast: https://api.openweathermap.org/data/2.5/forecast?xxxx&lat=xxxx&lon=-xxxx&units=imperial&lang=en&r=0.20717837289492247
2024-01-05 15:35:41.037885 GMT (UTC+0000) - INFO: wxmetar: xxxx 051520Z AUTO /////KT 9999 FEW047/// 09/04 Q1008
2024-01-05 15:35:41.048743 GMT (UTC+0000) - Traceback (most recent call last):
2024-01-05 15:35:41.053612 GMT (UTC+0000) - File "/home/xxx/PiClock/Clock/PyQtPiClock.py", line 1305, in wxfinished_metar
2024-01-05 15:35:41.058505 GMT (UTC+0000) - ws = (Config.LWind + wd + ' ' + '%.1f' % (f.wind_speed.value('KMH')) + 'km/h')
2024-01-05 15:35:41.063354 GMT (UTC+0000) - ^^^^^^^^^^^^^^^^^^
2024-01-05 15:35:41.068566 GMT (UTC+0000) - AttributeError: 'NoneType' object has no attribute 'value'

and the second attempt....

2024-01-05 15:39:27.704910 GMT (UTC+0000) - INFO: map base url for radar1: http://maps.googleapis.com/maps/api/staticmap?key=xxxx&center=xxxx,-xxxx&zoom=7&size=400x330&maptype=hybrid
2024-01-05 15:39:27.709065 GMT (UTC+0000) - INFO: map base url for radar2: http://maps.googleapis.com/maps/api/staticmap?key=xxxx&center=xxxx,-xxxx&zoom=11&size=400x330&maptype=hybrid
2024-01-05 15:39:27.713351 GMT (UTC+0000) - INFO: map base url for radar3: http://maps.googleapis.com/maps/api/staticmap?key=xxxx&center=xxxx,-xxxx&zoom=6&size=466x420&maptype=hybrid
2024-01-05 15:39:27.717447 GMT (UTC+0000) - INFO: map base url for radar4: http://maps.googleapis.com/maps/api/staticmap?key=xxxx&center=xxxx,-xxxx&zoom=10&size=466x420&maptype=hybrid
2024-01-05 15:39:28.327080 GMT (UTC+0000) - libpng warning: iCCP: known incorrect sRGB profile
2024-01-05 15:39:30.924637 GMT (UTC+0000) - INFO: getting METAR current conditions: https://tgftp.nws.noaa.gov/data/observations/metar/stations/xxxx.TXT
2024-01-05 15:39:30.973125 GMT (UTC+0000) - INFO: getting OpenWeather One Call: https://api.openweathermap.org/data/2.5/onecall?appid=xxxx&lat=xxxx&lon=-xxxx&units=imperial&lang=en&r=0.6063000165373416
2024-01-05 15:39:30.978680 GMT (UTC+0000) - INFO: wxstart for radar1
2024-01-05 15:39:30.984801 GMT (UTC+0000) - INFO: wxstart for radar2
2024-01-05 15:39:31.141869 GMT (UTC+0000) - WARNING: Response from api.openweathermap.org: 401 - Invalid API key. Please see https://openweathermap.org/faq#error401 for more info.
2024-01-05 15:39:31.145967 GMT (UTC+0000) - WARNING: OpenWeather One Call failed...
2024-01-05 15:39:31.150447 GMT (UTC+0000) - WARNING: Falling back to separate OpenWeather calls for current weather conditions and forecast
2024-01-05 15:39:31.154228 GMT (UTC+0000) - INFO: getting OpenWeather forecast: https://api.openweathermap.org/data/2.5/forecast?appid=xxxx&lat=xxxx&lon=-xxxx&units=imperial&lang=en&r=0.3651581171587942
2024-01-05 15:39:32.281712 GMT (UTC+0000) - INFO: wxmetar: xxxx 051520Z AUTO /////KT 9999 FEW047/// 09/04 Q1008
2024-01-05 15:39:32.286829 GMT (UTC+0000) - Traceback (most recent call last):
2024-01-05 15:39:32.291409 GMT (UTC+0000) - File "/home/xxx/PiClock/Clock/PyQtPiClock.py", line 1305, in wxfinished_metar
2024-01-05 15:39:32.296092 GMT (UTC+0000) - ws = (Config.LWind + wd + ' ' + '%.1f' % (f.wind_speed.value('KMH')) + 'km/h')
2024-01-05 15:39:32.300933 GMT (UTC+0000) - ^^^^^^^^^^^^^^^^^^
2024-01-05 15:39:32.305714 GMT (UTC+0000) - AttributeError: 'NoneType' object has no attribute 'value'

I hope this can help narrow down the issue.
Up until yesterday, or possibly the day before. it's been working perfectly ever since you added the fix.
But now I have the crashing, again.

Presumably if I wait, as before, then when the METAR data changes I should be able to launch PiClock, again.

My apologies for the continuing issues,
With thanks, and best regards.

SerBrynden, 2024-01-05:
Thank you.  I did see your posts both here and in the previously closed issue.  Your METAR station definitely is weird.  `/////KT` is definitely not a valid wind speed.  I'll add some checks for this in the code and have PiClock ignore invalid wind speeds.

SerBrynden, 2024-01-05:
Should be fixed now.  Update your PiClock with `git pull` and try again.

Vindictor, 2024-01-05:
> Should be fixed now. Update your PiClock with `git pull` and try again.

Thank you really, very much, for all your time and effort.
Kind regards, and best wishes for a good weekend.
Vin.

Vindictor, 2024-01-07:
I just wanted to say thanks again to @SerBrynden for his help.

This morning I booted up my PiClock, and from then and for most of the day the wind speed/direction was missing from my METAR on screen data.... but.... better this than crashing to the desktop!

The thing updated around an hour ago, and the wind data is showing.

I assume, therefore, that the fix that was implemented worked.... and that my METAR station once again send nonsense data out, and while this was happening my PiClock just displayed a blank line in the METAR output.
Again, this is much preferred to a full on crash to desktop, and behaves more like the older N0bel version I ran previously.

Thanks again.

As a brief aside.... ever since the update of approx 3 weeks ago, the one to fix my initial METAR crashing issues, for some reason, when I boot my PiClock first thing in the morning, I always have one blank weather box in the openweathermap data.
I think it's the fourth box down, the one that shows the all day forecast for the current day.  The first three boxes show forecasts several hours apart, and below those is the full day forecast for today.
Every morning, only since that update of three weeks ago, this box is blank on start up.
It's not at all a big deal.  After 30-60 mins, whenever the forecast updates automatically, the data comes back all of the forecast boxes are filled with data.  I haven't read anyone else mention it, so perhaps it only affects me.
I'm just mentioning it as a small, not important issue, while I'm here.  It doesn't cause a crash, and it always corrects itself within an hour so.... if noone else is affected, never mind.  It's just strange how it only started after that first update to fix the METAR crash.

Again, if noone else is affected, then never mind.
Other than this very slight, and brief glitch, everything appears to be working great.

As a second aside, I've mentioned previously that I connected a large (18tb) USB3 HDD to this Pi4 (it's the reason I switched to a Pi4) seeing as this Pi is on all day.  I formatted the HDD to ext4, enabled Samba, MiniDLNA and a few other things, and it works GREAT as a cheap NAS.  I get full gigabit transfer speeds out of it, and MiniDLNA means I can stream media direct to TVs, bluray players and consoles around the house.   An invaluable second use for an always on PiClock.  I'm amazed and relived at the sustained full gigabit transfer speeds, even on large transfers.  Gotta love the Pi's.  Hope this inspires someone else to do the same, just be sure to use EXT4 and not NTFS, as the free NTFS driver is pretty slow on write speeds (about 30MB/s instead of 112MB/s).  (I had to use HD-IDLE to get my HDD to spin down after x minutes of inactivity, but this works for me)

All the best.

SerBrynden, 2024-01-08:
The blank forecast box might have something to do with the coordinates used for the weather location not falling within the clock's timezone (e.g. UTC).  This problem could be compounded if you're not subscribed to the OpenWeath OneCall API, which is still free for the first 1000 calls per day but requires a credit card in case there are overages.  The OneCall offers current weather conditions, daily forecasts, and hourly forecasts.  If you're not subscribed to OneCall, PiClock falls back to the basic OpenWeather "current conditions and 3-hour forecast for 5 days" API calls.  In order to do daily forecasts, PiClock combines the 3-hour forecasts for each day.  Anyway… I think there is a problem if your clock’s date/time is showing Saturday, for example, due to the timezone setting, but it’s actually Sunday where you’re physically located.  So, it thinks there’s no weather forecast data for that day because it was in the past.  Or visa versa.  I’m confused now and haven’t thought about this problem in awhile.

SerBrynden, 2024-01-08:
I also rememberd that for daily forecasts, PiClock consideres the "weather day" to be 6AM to 6AM next day, in the local timezone at the given coordinates.  If there are no 3-hour forecasts from OpenWeather that fall within that time window, the box will be blank.

Vindictor, 2024-01-08:
> I also rememberd that for daily forecasts, PiClock consideres the "weather day" to be 6AM to 6AM next day, in the local timezone at the given coordinates. If there are no 3-hour forecasts from OpenWeather that fall within that time window, the box will be blank.

Interesting, good to know.
Having said that, like I said this only started happening since the first "metar fix update" of approx 3 weeks ago.
But it really doesn't matter.

For interest, I just grabbed a screencap of my PiClock at the moment, via VNC on my PC.  This is UK time.  Also, interestingly, no wind data, again!  I had no wind data all day yesterday until the evening.  I guess my local metar station is sending out more nonsense.  Maybe it's just not windy enough! ;) 

Thanks for all your help, anyway.  I'm delighted to still have the thing running after all these years, and I do genuinely rely on the rainviewer windows, and even have my elderly parents call me to ask if it's going to rain soon ;)

Best wishes, and kind regards from the UK.

[image]

feh123, 2024-01-08:
I live near Cambridge UK and currently openweathermap says the wind is 17km/h but looking out from my window it looks more like zero! Supposed to get windier this afternoon but not by a lot.

Vindictor, 2024-01-08:
I'm down the bottom end of the country.  I blurred the map but... you can still see all the blue around me ;)
Obviously I'm on a very different METAR station to you.

I've just come in.... the forecast data is complete now, it always seems to complete after one update, but no longer starts up with a complete forecast (just the one empty box)
My wind speed is still blank, although the other METAR info has updated, and the temperature increased.
But this is still preferable to having the thing crash to the desktop in case of erroneous data.  I'm not sure if the log stores the METAR info if there's no crash.  I'll check....

OK, yes, this is the latest update, from the log it's 10 mins ago as I write this, the on screen info shows METAR was updated at 10:20am, approx half an hour ago

2024-01-08 10:40:09.525997 GMT (UTC+0000) - INFO: wxmetar: xxxx 081020Z AUTO /////KT 9999 OVC020/// 02/M01 Q1030

So that's what's coming through.  I guess the ///// is just "nothing", hence no info on the screen.... which again, is better than a crash!

The log is filled with these...... however, everything appears to be working.  It's sunny where I am, but off the coast I can see rain clouds passing by, so it's all working.

`2024-01-08 10:40:11.956730 GMT (UTC+0000) - ERROR: Response from rainviewer.com: <?xml version="1.0" encoding="iso-8859-1"?>
2024-01-08 10:40:11.960243 GMT (UTC+0000) - <!DOCTYPE html PUBLIC "-//W3C//DTD XHTML 1.0 Transitional//EN"
2024-01-08 10:40:11.963622 GMT (UTC+0000) - "http://www.w3.org/TR/xhtml1/DTD/xhtml1-transitional.dtd">
2024-01-08 10:40:11.966992 GMT (UTC+0000) - <html xmlns="http://www.w3.org/1999/xhtml" xml:lang="en" lang="en">
2024-01-08 10:40:11.970357 GMT (UTC+0000) - <head>
2024-01-08 10:40:11.973591 GMT (UTC+0000) - <title>404 - Map not found</title>
2024-01-08 10:40:11.976850 GMT (UTC+0000) - </head>
2024-01-08 10:40:11.980120 GMT (UTC+0000) - <body>
2024-01-08 10:40:11.983461 GMT (UTC+0000) - <h1>404 - Map not found</h1>
2024-01-08 10:40:11.986630 GMT (UTC+0000) - </body>
2024-01-08 10:40:11.989945 GMT (UTC+0000) - </html>
2024-01-08 10:40:11.993146 GMT (UTC+0000) - 
2024-01-08 10:40:11.996579 GMT (UTC+0000) - ERROR: Response from rainviewer.com: <?xml version="1.0" encoding="iso-8859-1"?>
2024-01-08 10:40:11.999990 GMT (UTC+0000) - <!DOCTYPE html PUBLIC "-//W3C//DTD XHTML 1.0 Transitional//EN"
2024-01-08 10:40:12.003346 GMT (UTC+0000) - "http://www.w3.org/TR/xhtml1/DTD/xhtml1-transitional.dtd">
2024-01-08 10:40:12.007789 GMT (UTC+0000) - <html xmlns="http://www.w3.org/1999/xhtml" xml:lang="en" lang="en">
2024-01-08 10:40:12.011871 GMT (UTC+0000) - <head>
2024-01-08 10:40:12.016393 GMT (UTC+0000) - <title>404 - Map not found</title>
2024-01-08 10:40:12.021223 GMT (UTC+0000) - </head>
2024-01-08 10:40:12.024914 GMT (UTC+0000) - <body>
2024-01-08 10:40:12.028214 GMT (UTC+0000) - <h1>404 - Map not found</h1>
2024-01-08 10:40:12.031548 GMT (UTC+0000) - </body>
2024-01-08 10:40:12.034704 GMT (UTC+0000) - </html>
2024-01-08 10:40:12.037885 GMT (UTC+0000) - `

feh123, 2024-01-08:
My closest is Cambridge Airport. I had some questions about Metar - where do get your api key? Does Metar supply forecasts?
Thanks (and thanks to @SerBrynden for all his work on PiClock).

Vindictor, 2024-01-08:
Damn, I just glanced up to check the time as I need to go out.
The PiClock has crashed/closed, and the Pi is displaying the desktop.

I'll go into the log.... yep.... as expected:

2024-01-08 11:40:13.060402 GMT (UTC+0000) - 
2024-01-08 11:40:18.613181 GMT (UTC+0000) - INFO: getting METAR current conditions: https://tgftp.nws.noaa.gov/data/observations/metar/stations/xxxx.TXT
2024-01-08 11:40:18.622828 GMT (UTC+0000) - INFO: getting OpenWeather forecast: https://api.openweathermap.org/data/2.5/forecast?appid=xxx&lat=xxx&lon=-xxx&units=imperial&lang=en&r=0.6614221780934258
2024-01-08 11:40:19.890526 GMT (UTC+0000) - INFO: wxmetar: xx081130Z 04005KT 9999 FEW020 SCT30 03/MS01 Q1030 BLU TEMPO SCT024 RMK BLU WHT
2024-01-08 11:40:19.899207 GMT (UTC+0000) - /home/xxx/PiClock/venv/lib/python3.11/site-packages/metar/Metar.py:502: RuntimeWarning: Unparsed groups in body '03/MS01' while processing 'xxx 081130Z 04005KT 9999 FEW020 SCT30 03/MS01 Q1030 BLU TEMPO SCT024 RMK BLU WHT'
2024-01-08 11:40:19.905883 GMT (UTC+0000) - warnings.warn(message, RuntimeWarning)
2024-01-08 11:40:19.917811 GMT (UTC+0000) - Traceback (most recent call last):
2024-01-08 11:40:19.925044 GMT (UTC+0000) - File "/home/xxx/PiClock/Clock/PyQtPiClock.py", line 1307, in wxfinished_metar
2024-01-08 11:40:19.931814 GMT (UTC+0000) - temper.setText('%.1f' % (f.temp.value('C')) + u'Â°C')
2024-01-08 11:40:19.938263 GMT (UTC+0000) - ^^^^^^^^^^^^
2024-01-08 11:40:19.945203 GMT (UTC+0000) - AttributeError: 'NoneType' object has no attribute 'value'

Vindictor, 2024-01-08:
> My closest is Cambridge Airport. I had some questions about Metar - where do get your api key? Does Metar supply forecasts? Thanks (and thanks to @SerBrynden for all his work on PiClock).

You can find your metar code somewhere like here, where I've already looked up Cambridge Airport

https://metar-taf.com/?c=516985.1758.6&hl=EGSC

There is no API key.  Looks like your METAR code is EGSC.
If you go into the PiClock config file (I think it was config.py) then you'll find, commented out
metar = ' '

so change that to metar = 'EGSC' and when you restart PiClock, it will display the current conditions from your local metar station, which should be more accurate for your immediate area (assuming it doesn't cause your PiClock to crash, like mine does!)

Also remember to remove the # before the metar command in the config.

When it works, below the current conditions on the top left, you should see your metar code.
Have fun!

feh123, 2024-01-08:
Thanks - when I feel brave I will try it!

Vindictor, 2024-01-08:
> Thanks - when I feel brave I will try it!

It's no big deal
If you set a METAR code then the PiClock will use that instead of Openweather, or whatever.
If you put a # back in front of the metar = 'EGSC' command, then this will disable it and go back to whatever it's currently doing.

If in doubt, you can always make a backup copy of your config.py, first.
But you can literally just edit the command that's already there.  Just add EGSC between the ' ' and remove the # 
If you have issues, just place a # back in front of the word METAR again, save, and relaunch / reboot.

I don't think most people have these issues.  It's only me crying because my PiClock keeps crashing!
My local METAR is a military base, not an airport, so I've no idea what info they're sending out to pilots!
Anyway, I've powered down my Pi now, as I can't get PiClock to launch.

By the time I get back the METAR info will have (hopefully) updated again, and all will be fine.
Sorry to @SerBrynden for my continued aggravations!  Still, any METAR station could send out malformed info at some stage, so better to have the PiClock gracefully display no info, rather than completely crash to the desktop, in my opinion.

SerBrynden, 2024-01-08:
@feh123 METAR can only provide current conditions.  The forecasts will still come from OpenWeather or Tomorrow.io.

Vindictor, 2024-01-08:
Ah yes, sorry if I wasn't clear about METAR providing current conditions.
It's very convenient for me, as I live right on the coast and the conditions from other sources can be quite inaccurate.  
The METAR station is very close to my home, and I consider this METAR info to be the accurate data, ahead of other weather apps / sources.  

I got home approx half an hour ago, 1pm UK time. I booted up the Pi and everything started perfectly. 
The PiClock showed full METAR info, and no blank box on the right hand side, for the todays general forecast, which backs up what @SerBrynden said earlier.  

And now, just as I was opening the link to this page, the PiClock crashed to desktop.  I saw it happen. 
I'll log in and grab the log......

Yep, it updated 5 mins ago now, as I type this.....

2024-01-08 13:32:29.941298 GMT (UTC+0000) - INFO: getting METAR current conditions: https://tgftp.nws.noaa.gov/data/observations/metar/stations/xxxx.TXT
2024-01-08 13:32:29.946133 GMT (UTC+0000) - INFO: getting OpenWeather forecast: https://api.openweathermap.org/data/2.5/forecast?appid=xxxx&lat=xxxx&lon=-xxxx&units=imperial&lang=en&r=0.916737546284286
2024-01-08 13:32:30.532767 GMT (UTC+0000) - INFO: wxmetar: xxxx 081320Z 04015KT 9999 FEW025 04/MS01 Q1029 BLU TEMPO SCT024 RMK BLU WHT
2024-01-08 13:32:30.540277 GMT (UTC+0000) - /home/xxx/PiClock/venv/lib/python3.11/site-packages/metar/Metar.py:502: RuntimeWarning: Unparsed groups in body '04/MS01' while processing 'xxxx 081320Z 04015KT 9999 FEW025 04/MS01 Q1029 BLU TEMPO SCT024 RMK BLU WHT'
2024-01-08 13:32:30.544294 GMT (UTC+0000) - warnings.warn(message, RuntimeWarning)
2024-01-08 13:32:30.548838 GMT (UTC+0000) - Traceback (most recent call last):
2024-01-08 13:32:30.553197 GMT (UTC+0000) - File "/home/xxx/PiClock/Clock/PyQtPiClock.py", line 1307, in wxfinished_metar
2024-01-08 13:32:30.558023 GMT (UTC+0000) - temper.setText('%.1f' % (f.temp.value('C')) + u'°C')
2024-01-08 13:32:30.562488 GMT (UTC+0000) - ^^^^^^^^^^^^
2024-01-08 13:32:30.567267 GMT (UTC+0000) - AttributeError: 'NoneType' object has no attribute 'value'

Vindictor, 2024-01-08:
@SerBrynden I guess it doesn't really matter.  
If it helps, here is my metar station info.  
https://metar-taf.com/EGDR

feh123, 2024-01-08:
> @feh123 METAR can only provide current conditions. The forecasts will still come from OpenWeather or Tomorrow.io.

Ah. Okay, thanks. So the top LHS box will be METAR and RHS column will be my current provide OWM? That's very nice.

Vindictor, 2024-01-08:
> > @feh123 METAR can only provide current conditions. The forecasts will still come from OpenWeather or Tomorrow.io.
> 
> Ah. Okay, thanks. So the top LHS box will be METAR and RHS column will be my current provide OWM? That's very nice.

Yes, sorry.  I thought I'd mentioned that the metar affects the top left hand side of the screen.  so all the current weather info comes from a local airport/weather station, and can be much more accurate than a weather service.  Certainly in my case, anyway (when it works properly!)

In your case, I guess you could just click here, and see the current weather conditions that will be sent to your PiClock, if you decide to try METAR.

https://metar-taf.com/EGSC

feh123, 2024-01-08:
Thanks. I guess I could risk just adding METAR to my current PiClock? Or is that not a good idea?

Vindictor, 2024-01-08:
> Thanks. I guess I could risk just adding METAR to my current PiClock? Or is that not a good idea?

It's no risk.  give it a try and you may get more accurate current conditions. 
If you don't like it for some reason, or find that it doesn't work, you can just comment it out again.  

ie. in the Config.py file you already have an entry called something like

#METAR = ' '

You add your METAR code between ' ' and remove the # 
the # means that something is ignored, or that it's just a "comment"

so if you have an issue, just go back to Config.py, find the METAR part again, and add a # back at the beginning. 

Or as I said above, if you're afraid of breaking something, you can always backup the Config.py file first...  which you could always copy back later, if you needed to.  

One other thing I've noticed... while I have METAR disabled here, it looks as though the current conditions text is **BOLD**

Normally (with METAR) the text looks a lot.... thinner!
Hmmm, it's actually easier to read as it currently is, **with bold text!**

Anyway, have fun. 
All you have to do is add that # back before the word METAR in your Config.py and it'll be disabled, again. (No need to edit out your metar code, once you've added it, it'll just be completely ignored when # is in front of it)

Hope this makes sense, and helps.

feh123, 2024-01-08:
I do hit a problem - I updated to RPi Bookworm and re-installed PiClock 17-Nov-23. But I did add back my config.py and some images to get back to my setup. So there is no METAR in my config.py, also there is no METAR in my PyQtPiClock.py. For the PtQtPiClock.py this is odd because according to my notes I did not touch this. 
I have all my files in a backup so no problem doing it - just need a bit of time.

SerBrynden, 2024-01-08:
Create a new config file based on the latest config example file found in the folder.  Things have changed if you haven't done that in awhile.

feh123, 2024-01-08:
Thanks @SerBrynden I have downloaded the master.zip and will add the new config.py and PyQtPiClock.py back to my PiClock.

SerBrynden, 2024-01-08:
Why not `git pull` and run update.sh?  Or backup your images and `git clone` the whole thing to start from scratch?  There may be updates to other files as well.  If you use `git`, future updates will be easier and updating shouldn't touch any images you added.

feh123, 2024-01-08:
Okay - I will do a git pull. Thanks!

feh123, 2024-01-09:
Okay - the git pull worked well. I needed to add back my images and settings to the config-example.py then rename and replace the current config.py with this file. I used EGSS as it had a nice weather icon too. It's also more relevant because Cambridge is really the fens and I am closer to the hills (the eastern end of the Chiltens). Not hills like a US hill of course!
Looks good and no crashes so far.
Thanks for the help.

feh123, 2024-01-10:
Just to say my PiClock is fine. I am using EGSS (Stansted). I also see that sometimes the weather logo is missing but then it returns usually at the next update - which seems to be 30 mins.

Vindictor, 2024-01-10:
> Just to say my PiClock is fine. I am using EGSS (Stansted). I also see that sometimes the weather logo is missing but then it returns usually at the next update - which seems to be 30 mins.

Great stuff :)
Yes, that's how mine used to behave.
The issue is clearly with my metar station.
If you changed your metar to the same as mine for a test, EGDR, you'd probably get a crash as well.
It's just the crashing that's the biggest issue for me.  If data is missing due to a metar error, then absolutely fair enough.
But obviously when the app crashes to the PiOS desktop then it needs manual restarting, which will also fail until the metar data has updated again.

For now I shall comment out metar again from my config file, and change it back, later.
My apologies to @SerBrynden if I'm sounding like a whining user.
Still, if my local metar station puts out bad data from time to time, I suppose any metar station could, and so having the PiClock deal with it without crashing is a worthy update to make.... easy for me to say, of course, I'm not doing the work!

SerBrynden, 2024-02-06:
@Vindictor PiClock has been updated to handle more of those METAR errors.  Do `git pull` to update.  I still don't know why you get the blank daily forecast box sometimes.  If you keep having that problem, please send more screenshots, note the time it's happening, and any errors or strangeness from the log file.

closed 2024-02-06 by SerBrynden

Vindictor, 2024-02-06:
> @Vindictor PiClock has been updated to handle more of those METAR errors. Do `git pull` to update. I still don't know why you get the blank daily forecast box sometimes. If you keep having that [...]

Hello mate,
Thanks very much for all your effort.  
My PiClock with METAR hasn't actually crashed in a while, now.  I've had the occasion where something was missing from the METAR info, but no crashes.  most recently I recall simply having the icon above, the one that shows the current weather conditions, being blank for a short while.  But no crashes!

Yes, I still have the blank daily forecast box.  Not just sometimes, but every single morning on first start.  This never, ever used to happen, until the first update you made (from the previous thread) to help with the METAR issue.  Maybe a coincidence, but it's done it every morning since then, and fixes itself with a subsequent update, maybe even the next one. It doesn't stay blank for too long.  I don't really worry about it, as long as it doesn't crash. 
Maybe previously, if this data wasn't available online, it would pull from a cache, and that's been changed?  I've no idea. 
As I said above, after your message regarding the 6am - 6pm time zone, this would appear to make sense as when I rebooted my PiClock later in the day, it started up with all the forecast boxes full. 
I generally boot it somewhere in the region of 9:30am, UK time. 
It's really no big deal.  The data fills in a little later, and the main thing is no crash to desktop, which requires a manual restart. 

Thank you once again for all your support.  
I'll perform a git pull and update before I shut it down for the night. 

Kind regards,
Vin.

Vindictor, 2024-02-07:
Hi there, again.

So this morning I was paying attention to the clock, as best I could, while doing other things.
I was late booting it up, 09:53am (UK) and the forecast box was still empty.

I kept my eye on it, until I eventually decided to VNC in and grab a screencap and check the log.
I saw nothing in the log, out of the ordinary, no error.

But here is the screen cap

[image]

I noticed that the METAR info hadn't updated since 09:20, so was keeping an eye out for when that updated, to see if the missing weather box would also fill in.
I didn't have to wait long, just a few minutes later, it updated, you can see the time it updated, and my local time from the screen cap, and all weather boxes are full.

[image]

So this is what happens every morning.

While I'm here, I guess it wouldn't hurt to paste a log in below.
This is starting from the first entry of todays date.
Interestingly, I don't even see a METAR entry until the 10:23am update.  But anyway, it's all working, and looking great at the moment :)

```
2024-02-07 09:53:56.115611 GMT (UTC+0000) - INFO: radar1... get radar tiles for time 1707296400 (2024-02-07 09:00:00+00:00)
2024-02-07 09:53:56.118263 GMT (UTC+0000) - INFO: radar1 tile0 https://tilecache.rainviewer.com/v2/radar/1707296400//256/7/61/42/6/1_1.png
2024-02-07 09:53:56.120917 GMT (UTC+0000) - INFO: radar2... get radar tiles for time 1707296400 (2024-02-07 09:00:00+00:00)
2024-02-07 09:53:56.123853 GMT (UTC+0000) - INFO: radar2 tile0 https://tilecache.rainviewer.com/v2/radar/1707296400//256/11/993/693/6/1_1.png
2024-02-07 09:53:56.152284 GMT (UTC+0000) - INFO: radar1 tile1 https://tilecache.rainviewer.com/v2/radar/1707296400//256/7/62/42/6/1_1.png
2024-02-07 09:53:56.190227 GMT (UTC+0000) - INFO: radar2 tile1 https://tilecache.rainviewer.com/v2/radar/1707296400//256/11/994/693/6/1_1.png
2024-02-07 09:53:56.223598 GMT (UTC+0000) - INFO: radar2 tile2 https://tilecache.rainviewer.com/v2/radar/1707296400//256/11/993/694/6/1_1.png
2024-02-07 09:53:56.275958 GMT (UTC+0000) - INFO: radar1 tile2 https://tilecache.rainviewer.com/v2/radar/1707296400//256/7/61/43/6/1_1.png
2024-02-07 09:53:56.280501 GMT (UTC+0000) - INFO: radar2 tile3 https://tilecache.rainviewer.com/v2/radar/1707296400//256/11/994/694/6/1_1.png
2024-02-07 09:53:56.310013 GMT (UTC+0000) - INFO: radar1 tile3 https://tilecache.rainviewer.com/v2/radar/1707296400//256/7/62/43/6/1_1.png
2024-02-07 09:53:56.318611 GMT (UTC+0000) - INFO: radar2... get radar tiles for time 1707296400 (2024-02-07 09:00:00+00:00)
2024-02-07 09:53:56.321264 GMT (UTC+0000) - INFO: radar2... get radar tiles for time 1707297000 (2024-02-07 09:10:00+00:00)
2024-02-07 09:53:56.323910 GMT (UTC+0000) - INFO: radar2 tile0 https://tilecache.rainviewer.com/v2/radar/1707297000//256/11/993/693/6/1_1.png
2024-02-07 09:53:56.347870 GMT (UTC+0000) - INFO: radar2 tile1 https://tilecache.rainviewer.com/v2/radar/1707297000//256/11/994/693/6/1_1.png
2024-02-07 09:53:56.359078 GMT (UTC+0000) - INFO: radar1... get radar tiles for time 1707296400 (2024-02-07 09:00:00+00:00)
2024-02-07 09:53:56.361704 GMT (UTC+0000) - INFO: radar1... get radar tiles for time 1707297000 (2024-02-07 09:10:00+00:00)
2024-02-07 09:53:56.364314 GMT (UTC+0000) - INFO: radar1 tile0 https://tilecache.rainviewer.com/v2/radar/1707297000//256/7/61/42/6/1_1.png
2024-02-07 09:53:56.380135 GMT (UTC+0000) - INFO: radar2 tile2 https://tilecache.rainviewer.com/v2/radar/1707297000//256/11/993/694/6/1_1.png
2024-02-07 09:53:56.395248 GMT (UTC+0000) - INFO: radar1 tile1 https://tilecache.rainviewer.com/v2/radar/1707297000//256/7/62/42/6/1_1.png
2024-02-07 09:53:56.414044 GMT (UTC+0000) - INFO: radar2 tile3 https://tilecache.rainviewer.com/v2/radar/1707297000//256/11/994/694/6/1_1.png
2024-02-07 09:53:56.430725 GMT (UTC+0000) - INFO: radar1 tile2 https://tilecache.rainviewer.com/v2/radar/1707297000//256/7/61/43/6/1_1.png
2024-02-07 09:53:56.449748 GMT (UTC+0000) - INFO: radar2... get radar tiles for time 1707296400 (2024-02-07 09:00:00+00:00)
2024-02-07 09:53:56.452443 GMT (UTC+0000) - INFO: radar2... get radar tiles for time 1707297000 (2024-02-07 09:10:00+00:00)
2024-02-07 09:53:56.455331 GMT (UTC+0000) - INFO: radar2... get radar tiles for time 1707297600 (2024-02-07 09:20:00+00:00)
2024-02-07 09:53:56.458333 GMT (UTC+0000) - INFO: radar2 tile0 https://tilecache.rainviewer.com/v2/radar/1707297600//256/11/993/693/6/1_1.png
2024-02-07 09:53:56.467241 GMT (UTC+0000) - INFO: radar1 tile3 https://tilecache.rainviewer.com/v2/radar/1707297000//256/7/62/43/6/1_1.png
2024-02-07 09:53:56.480837 GMT (UTC+0000) - INFO: radar2 tile1 https://tilecache.rainviewer.com/v2/radar/1707297600//256/11/994/693/6/1_1.png
2024-02-07 09:53:56.524939 GMT (UTC+0000) - INFO: radar1... get radar tiles for time 1707296400 (2024-02-07 09:00:00+00:00)
2024-02-07 09:53:56.528208 GMT (UTC+0000) - INFO: radar1... get radar tiles for time 1707297000 (2024-02-07 09:10:00+00:00)
2024-02-07 09:53:56.531050 GMT (UTC+0000) - INFO: radar1... get radar tiles for time 1707297600 (2024-02-07 09:20:00+00:00)
2024-02-07 09:53:56.536167 GMT (UTC+0000) - INFO: radar1 tile0 https://tilecache.rainviewer.com/v2/radar/1707297600//256/7/61/42/6/1_1.png
2024-02-07 09:53:56.542735 GMT (UTC+0000) - INFO: radar2 tile2 https://tilecache.rainviewer.com/v2/radar/1707297600//256/11/993/694/6/1_1.png
2024-02-07 09:53:56.562542 GMT (UTC+0000) - INFO: radar1 tile1 https://tilecache.rainviewer.com/v2/radar/1707297600//256/7/62/42/6/1_1.png
[... cut ...]
2024-02-07 10:23:57.068249 GMT (UTC+0000) - INFO: radar1... get radar tiles for time 1707299400 (2024-02-07 09:50:00+00:00)
2024-02-07 10:23:57.072548 GMT (UTC+0000) - INFO: radar1... get radar tiles for time 1707300000 (2024-02-07 10:00:00+00:00)
2024-02-07 10:23:57.076434 GMT (UTC+0000) - INFO: radar1... get radar tiles for time 1707300600 (2024-02-07 10:10:00+00:00)
2024-02-07 10:23:57.080360 GMT (UTC+0000) - INFO: radar1... get radar tiles for time 1707301200 (2024-02-07 10:20:00+00:00)
```