SerBrynden/PiClock #13: Not sure what to call this.

maserowik, 2024-10-27:
I had to reload my Pi-Clock due to sd failure and noted a few issues that got me confused.

On my version of PI software the config.txt file moved to 
/boot/firmware/config.txt

So now when I have the program running my screen goes blank with in 10 seconds.
What I had to do was go login in via vnc and disconnect to keep screen active.

I did find this article and so far it has been working for me.
https://pimylifeup.com/raspberry-pi-disable-screen-blanking/

Here is my os setup

pi@pi-clock:~ $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 12 (bookworm)"
NAME="Raspbian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

Mike- N3BSQ

SerBrynden, 2024-10-28:
Is there a question here or a problem specifically with PiClock?  
I'm running the same version of Raspberry Pi OS (bookworm) and never experienced screen blanking, even though screen blanking is still enabled in my Raspberry Pi configuration settings.

maserowik, 2024-10-28:
Ok not sure why I had to change it on my end.  It works so I guess it a non issue

I close the issue later on tonight

Sent from the all new AOL app for iOS

closed 2024-10-28 by maserowik