SerBrynden/PiClock #15: No Weather Display Or Info

godofbooks, 2024-11-27:
Howdy, reinstalled per recommendation in https://github.com/n0bel/PiClock/issues/279. Still having the same issues. I've attached the details of the log file below:

2024-11-27 13:55:00.306577 MST (UTC-0700) - QStandardPaths: wrong permissions on runtime directory /run/user/1000, 0770 instead of 0700
2024-11-27 13:55:01.919986 MST (UTC-0700) - libpng warning: iCCP: known incorrect sRGB profile
2024-11-27 13:55:01.963115 MST (UTC-0700) - libpng warning: iCCP: known incorrect sRGB profile
2024-11-27 13:55:01.990128 MST (UTC-0700) - libpng warning: iCCP: known incorrect sRGB profile
2024-11-27 13:55:02.025966 MST (UTC-0700) - INFO: map base url for radar1: https://api.mapbox.com/styles/v1/bcurley/cj712peyz0bwr2sqfndbggupb/static/-105.090408,39.803261,6,0,0/400x330?access_token=[API KEY]&attribution=false&logo=false
2024-11-27 13:55:02.056816 MST (UTC-0700) - INFO: map overlay url for radar1: https://api.mapbox.com/styles/v1/bcurley/cj712r01c0bw62rm9isme3j8c/static/-105.090408,39.803261,6,0,0/400x330?access_token=[API KEY]
2024-11-27 13:55:02.103882 MST (UTC-0700) - INFO: map base url for radar2: https://api.mapbox.com/styles/v1/bcurley/cj712peyz0bwr2sqfndbggupb/static/-105.090408,39.803261,10,0,0/400x330?access_token=[API KEY]&attribution=false&logo=false
2024-11-27 13:55:02.135854 MST (UTC-0700) - INFO: map overlay url for radar2: https://api.mapbox.com/styles/v1/bcurley/cj712r01c0bw62rm9isme3j8c/static/-105.090408,39.803261,10,0,0/400x330?access_token=[API KEY]
2024-11-27 13:55:02.175278 MST (UTC-0700) - INFO: map base url for radar3: https://api.mapbox.com/styles/v1/bcurley/cj712peyz0bwr2sqfndbggupb/static/-105.090408,39.803261,6,0,0/933x840?access_token=[API KEY]&attribution=false&logo=false
2024-11-27 13:55:02.206124 MST (UTC-0700) - INFO: map overlay url for radar3: https://api.mapbox.com/styles/v1/bcurley/cj712r01c0bw62rm9isme3j8c/static/-105.090408,39.803261,6,0,0/933x840?access_token=[API KEY]
2024-11-27 13:55:02.241545 MST (UTC-0700) - INFO: map base url for radar4: https://api.mapbox.com/styles/v1/bcurley/cj712peyz0bwr2sqfndbggupb/static/-105.090408,39.803261,10,0,0/933x840?access_token=[API KEY]&attribution=false&logo=false
2024-11-27 13:55:02.268942 MST (UTC-0700) - INFO: map overlay url for radar4: https://api.mapbox.com/styles/v1/bcurley/cj712r01c0bw62rm9isme3j8c/static/-105.090408,39.803261,10,0,0/933x840?access_token=[API KEY]
2024-11-27 13:55:10.657301 MST (UTC-0700) - libpng warning: iCCP: known incorrect sRGB profile
2024-11-27 13:55:17.369110 MST (UTC-0700) - qt.qpa.wayland: Wayland does not support QWindow::requestActivate()
2024-11-27 13:55:41.128915 MST (UTC-0700) - INFO: getting METAR current conditions: https://tgftp.nws.noaa.gov/data/observations/metar/stations/DEN.TXT
2024-11-27 13:55:43.798637 MST (UTC-0700) - INFO: getting OpenWeather One Call: https://api.openweathermap.org/data/2.5/onecall?appid=[API KEY]&lat=39.803261&lon=-105.090408&units=imperial&lang=en&r=0.9968406647874285
2024-11-27 13:55:43.938690 MST (UTC-0700) - INFO: wxstart for radar1
2024-11-27 13:55:43.992570 MST (UTC-0700) - INFO: wxstart for radar2
2024-11-27 13:55:44.465725 MST (UTC-0700) - WARNING: Traceback (most recent call last):
2024-11-27 13:55:44.502159 MST (UTC-0700) - File "/home/piclock/PiClock/Clock/PyQtPiClock.py", line 423, in wxfinished_owm_onecall
2024-11-27 13:55:44.532956 MST (UTC-0700) - wxdata = json.loads(wxstr)
2024-11-27 13:55:44.557950 MST (UTC-0700) - ^^^^^^^^^^^^^^^^^
2024-11-27 13:55:44.607880 MST (UTC-0700) - File "/usr/lib/python3.11/json/__init__.py", line 346, in loads
2024-11-27 13:55:44.639427 MST (UTC-0700) - return _default_decoder.decode(s)
2024-11-27 13:55:44.691881 MST (UTC-0700) - ^^^^^^^^^^^^^^^^^^^^^^^^^^
2024-11-27 13:55:44.735895 MST (UTC-0700) - File "/usr/lib/python3.11/json/decoder.py", line 337, in decode
2024-11-27 13:55:44.774935 MST (UTC-0700) - obj, end = self.raw_decode(s, idx=_w(s, 0).end())
2024-11-27 13:55:44.801295 MST (UTC-0700) - ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
2024-11-27 13:55:44.831994 MST (UTC-0700) - File "/usr/lib/python3.11/json/decoder.py", line 355, in raw_decode
2024-11-27 13:55:44.865396 MST (UTC-0700) - raise JSONDecodeError("Expecting value", s, err.value) from None
2024-11-27 13:55:44.900105 MST (UTC-0700) - json.decoder.JSONDecodeError: Expecting value: line 1 column 1 (char 0)
2024-11-27 13:55:44.931451 MST (UTC-0700) - 
2024-11-27 13:55:44.958921 MST (UTC-0700) - WARNING: Response from api.openweathermap.org:
2024-11-27 13:55:44.995816 MST (UTC-0700) - WARNING: Moving on...
2024-11-27 13:55:45.025202 MST (UTC-0700) - ERROR: Response from nws.noaa.gov:
2024-11-27 13:55:45.061575 MST (UTC-0700) - ERROR: Response from api.mapbox.com:
2024-11-27 13:55:45.078071 MST (UTC-0700) - ERROR: Response from api.mapbox.com:
2024-11-27 13:55:45.115270 MST (UTC-0700) - ERROR: Response from api.mapbox.com:
2024-11-27 13:55:45.146544 MST (UTC-0700) - ERROR: Response from api.mapbox.com:
2024-11-27 13:55:45.182008 MST (UTC-0700) - ERROR: Response from api.mapbox.com:
2024-11-27 13:55:45.198684 MST (UTC-0700) - ERROR: Response from api.mapbox.com:
2024-11-27 13:55:45.235323 MST (UTC-0700) - ERROR: Response from api.mapbox.com:
2024-11-27 13:55:45.265254 MST (UTC-0700) - ERROR: Response from api.mapbox.com:
2024-11-27 13:55:53.110573 MST (UTC-0700) - INFO: radar1... get radar tiles for time 1732737600 (2024-11-27 13:00:00-07:00)
2024-11-27 13:55:53.195340 MST (UTC-0700) - INFO: radar1 tile0 https://tilecache.rainviewer.com/v2/radar/1732737600//256/7/25/47/6/1_1.png
2024-11-27 13:55:53.363097 MST (UTC-0700) - INFO: radar2... get radar tiles for time 1732737600 (2024-11-27 13:00:00-07:00)
2024-11-27 13:55:53.418347 MST (UTC-0700) - INFO: radar2 tile0 https://tilecache.rainviewer.com/v2/radar/1732737600//256/11/425/776/6/1_1.png
2024-11-27 13:55:53.481881 MST (UTC-0700) - ERROR: Response from rainviewer.com:
2024-11-27 13:55:53.539882 MST (UTC-0700) - ERROR: Response from rainviewer.com:

Any help or recommendations are much appreciated.

SerBrynden, 2024-11-27:
Is it connected to the internet?  That's a lot of errors and looks like it's getting no response from the various websites.

godofbooks, 2024-12-01:
Did another fresh install- you were right! Everything is working now. For some reason, my pi didn't have the proper network config set up. Fixed that and everything worked. Thanks for your feedback!

closed 2024-12-01 by godofbooks