SerBrynden/PiClock #17: [BUG]

maserowik, 2025-01-09:
I had to redo my Pi-Clock setup due to sd card going bad.   so i am now getting this message when I issue a bash startup.sh

INFO: wxstart for radar1
INFO: wxstart for radar2
INFO: wxmetar: KAGC 091953Z 33008G16KT 10SM CLR M05/M14 A3017 RMK AO2 SLP240 T10501144
127.0.0.1 - - [09/Jan/2025 15:11:01] "GET /temp HTTP/1.1" 200 -
Traceback (most recent call last):
  File "/home/pi/PiClock/Clock/PyQtPiClock.py", line 186, in tick
    float(clockrect.width()) / ts.height(),
    ~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~
ZeroDivisionError: float division by zero
startup.sh: line 119:  1665 Aborted                 python3 -u PyQtPiClock.py

i redid the setup and same error anyone have any suggestions?

Here is my current setup.

RPiZero 2 W Rev 1.0 (1GB) : Debian GNU/Linux 12 (bookworm) 64-bit (arm64) : Desktop Version : installed on 2025-01-08 19:29:03

SerBrynden, 2025-01-09:
You installed `python3-pyqt5`?
Does `sechand.png` exist under your `PiClock/Clock/images` folder?

maserowik, 2025-01-09:
> You installed `python3-pyqt5`? Does `sechand.png` exist under your `PiClock/Clock/images` folder?

pi@pi-clock:~ $ sudo apt install python3-pyqt5
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
python3-pyqt5 is already the newest version (5.15.9+dfsg-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
pi@pi-clock:~ $ ls
Bookshelf  Desktop  Documents  Downloads  Music  PiClock  Pictures  Public  Templates  Videos
pi@pi-clock:~ $ cd PiClock/Clock/images/
pi@pi-clock:~/PiClock/Clock/images $ ls second.png
ls: cannot access 'second.png': No such file or directory
pi@pi-clock:~/PiClock/Clock/images $ ls
berlin-at-night-mrwallpaper.jpg  clockbackground-kevin.png  clockface3.png          minhand-darkblue.png   sechand-darkgreen.png  squares1-jean.png   squares2-kevin.png  xenon462_squares1-jean.png
clockbackground-chris.png        clockbackground.png        hourhand-darkblue.png   minhand-darkgreen.png  sechand.png            squares1-kevin.png  xenon462_hour.png   xenon462_squares2-jean.png
clockbackground-jean.png         clockface3-darkblue.png    hourhand-darkgreen.png  minhand.png            slideshow              squares2-green.png  xenon462_min.png    xenon462_texture.png
clockbackground-kelly.png        clockface3-darkgreen.png   hourhand.png            sechand-darkblue.png   squares1-green.png     squares2-jean.png   xenon462_sec.png    xenon462_tsiferblat.png
pi@pi-clock:~/PiClock/Clock/images $

SerBrynden, 2025-01-09:
I'm not sure.  You had PiClock running on this same RPiZero 2 W before?

maserowik, 2025-01-09:
It was in 32 bit bullseye rpi 2 w originalNow 64 bit bookworm  rpi version 2 I am going to reformat and got to bullseye 64Michael A. SerowikSent from my iPhone

SerBrynden, 2025-01-10:
It should work on 64-bit Bookworm.  That's what I'm running.  The instructions and config have changed since Bullseye, so make sure you're following the latest instructions below and no trying to re-use an old config file.
https://github.com/SerBrynden/PiClock/blob/master/Documentation/Install.md

maserowik, 2025-01-14:
Not quite sure what the issue was but i did the install on bulleye 32 and 64 bit still had the same error.   i then tried bookwarm 64 bit again and it works.   not sure what  was wrong but it appears to be working.   Closing

closed 2025-01-14 by maserowik