SerBrynden/PiClock #19: Rainviewer not responding in UK

feh123, 2025-09-10:
**Describe the problem**
A clear and concise description of your problem.

**Operating System**
Use command `hostnamectl` in Linux terminal window to find OS info.
 - OS: [e.g. Raspberry Pi OS, Debian, Ubuntu, Windows]
 - Version: [e.g. Bookworm, 12.0]

**Python Version**
Use command `python --version` or `python3 --version` in terminal window to find Python version.
- Version: [e.g. 3.9]

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Log Output**
Copy relevant portion of error log here. Remove sensitive API keys.

**Additional context**
Add any other context about the problem here.

Vindictor, 2025-09-16:
Hmmm, still not working. 
Out of interest I thought I'd recheck what updates were installed earlier today. 
Shouldn't really cause an issue (?) It's a 404 error so, I assume it's on their (rainviewer) side. 

I'm just surprised it hasn't started working yet, or that other people haven't had the same issue. 
Anyway, here are the updates from earlier today, applied to my Bookworm PiClock.... indeed, now I come to check again,
they're mostly related to VLC. 

`$ cat /var/log/dpkg.log | grep "\ upgrade\ " |grep "2025-09-16"
2025-09-16 15:42:13 upgrade vlc:arm64 1:3.0.21-0+rpt4+deb12u1 1:3.0.21-0+rpt5+deb12u1
2025-09-16 15:42:13 upgrade vlc-plugin-base:arm64 1:3.0.21-0+rpt4+deb12u1 1:3.0.21-0+rpt5+deb12u1
2025-09-16 15:42:17 upgrade vlc-data:all 1:3.0.21-0+rpt4+deb12u1 1:3.0.21-0+rpt5+deb12u1
2025-09-16 15:42:34 upgrade libvlccore9:arm64 1:3.0.21-0+rpt4+deb12u1 1:3.0.21-0+rpt5+deb12u1
2025-09-16 15:42:38 upgrade vlc-plugin-skins2:arm64 1:3.0.21-0+rpt4+deb12u1 1:3.0.21-0+rpt5+deb12u1
2025-09-16 15:42:39 upgrade vlc-plugin-qt:arm64 1:3.0.21-0+rpt4+deb12u1 1:3.0.21-0+rpt5+deb12u1
2025-09-16 15:42:39 upgrade vlc-plugin-video-output:arm64 1:3.0.21-0+rpt4+deb12u1 1:3.0.21-0+rpt5+deb12u1
2025-09-16 15:42:40 upgrade libvlc5:arm64 1:3.0.21-0+rpt4+deb12u1 1:3.0.21-0+rpt5+deb12u1
2025-09-16 15:42:40 upgrade vlc-bin:arm64 1:3.0.21-0+rpt4+deb12u1 1:3.0.21-0+rpt5+deb12u1
2025-09-16 15:42:41 upgrade libvlc-bin:arm64 1:3.0.21-0+rpt4+deb12u1 1:3.0.21-0+rpt5+deb12u1
2025-09-16 15:42:41 upgrade libwlroots-0.18:arm64 0.18.2-3+rpt3~bpo12+1 0.18.2-3+rpt4
2025-09-16 15:42:41 upgrade vlc-l10n:all 1:3.0.21-0+rpt4+deb12u1 1:3.0.21-0+rpt5+deb12u1
2025-09-16 15:42:50 upgrade vlc-plugin-access-extra:arm64 1:3.0.21-0+rpt4+deb12u1 1:3.0.21-0+rpt5+deb12u1
2025-09-16 15:42:53 upgrade vlc-plugin-notify:arm64 1:3.0.21-0+rpt4+deb12u1 1:3.0.21-0+rpt5+deb12u1
2025-09-16 15:42:54 upgrade vlc-plugin-samba:arm64 1:3.0.21-0+rpt4+deb12u1 1:3.0.21-0+rpt5+deb12u1
2025-09-16 15:42:54 upgrade vlc-plugin-video-splitter:arm64 1:3.0.21-0+rpt4+deb12u1 1:3.0.21-0+rpt5+deb12u1
2025-09-16 15:42:54 upgrade vlc-plugin-visualization:arm64 1:3.0.21-0+rpt4+deb12u1 1:3.0.21-0+rpt5+deb12u1`

feh123, 2025-09-16:
> Vindictor

Hi

My rainviewer stopped working sometime this afternoon and is still down. And yes when I updated the RPI OS most of the updates were for software installed like VLC.

I do have the rain viewer iOS app on my phone and that is working fine. But check out this blog from their web site: https://www.rainviewer.com/api.html

Sounds like the way PiClock interacts with Rainviewer needs to change although my understanding of this is zero so the why, what and what to do is well beyond me!

SerBrynden, 2025-09-16:
The problem is with rainviewer, not PiClock or Raspberry Pi.  Even with rainviewer's API transition, I don't see anything that needs to be changed for continued free non-commercial use.  I suspect they may be having problems with their servers during this transition.  Keep calm and carry on.

Vindictor, 2025-09-18:
> Okay, thanks for the fast response. I was rather unnerved by this idea of a transition!

Just out of interest, it's been a couple of days.  Has your rainviewer started working again, yet?
No luck, here.  I was unsure as to whether it's affecting all regions, or just the UK.  Maybe other users aren't having issues. 
Still keeping my fingers crossed that it returns "on its own" :) 
It's funny how reliant we become on such things.  I use mine to judge whether to do the washing that day, or go out without a jacket. Glance at my PiClock, check the rainviewer clouds (if any) and decide. 
Cheers.

feh123, 2025-09-18:
Sadly no. I found some more info on rainviewer's plans: https://www.rainviewer.com/api/transition-faq.html

Check out point 3 - I am not sure what PiClock uses but the single radar data API has been much restricted in September and much else has already gone. Point 4 talks about Patrons? 

I agree what is surprising is that more of us are not seeing the loss of radar. I can usually see the rainful for the UK and further away which is great for guessing my own forecast.

Vindictor, 2025-09-18:
Yes, I spotted that, earlier.
Rather than speculate, I'll just trust the post from @SerBrynden 
Point 3 suggests to me that it's regarding previous/historical radar images.

Having said that, point 4 only shows a zoom level limitation, and rate limiting.  Tiled radar data still available.
Then Jan 2026 even less zoom options, and a rate limit of 100 requests per minute, which shouldn't affect us.

I have no idea why it's ceased working.
But @SerBrynden surely does understand how the API works.  If no changes have been made on their end, to how the API works, then hopefully it is still an ongoing server issue causing the 404s, and service will return.  It's been a few days, but I'll actually be delighted if I suddenly see some rainclouds on the horizon!!

feh123, 2025-09-18:
Thankks this is great. Do I use git pull, I can't see the code in the png files to edit the code directly. Thanks again!