SerBrynden/PiClock #21: Trouble installing

GrampDamnpus, 2025-09-26:
While trying to setup and install PiClock Im running into an error which is likely caused by me but Im lost how to fix it. 

Raspbian bullseye 32

- Version: Python 3.9.2

(venv) pi@ShedClock:~/PiClock $ python3 -m pip install -r requirements.txt
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting python-dateutil>=2.8.2 (from -r requirements.txt (line 1))
  Downloading https://www.piwheels.org/simple/python-dateutil/python_dateutil-2.9.0.post0-py2.py3-none-any.whl (229 kB)
Collecting metar>=1.11.0 (from -r requirements.txt (line 2))
  Downloading https://www.piwheels.org/simple/metar/metar-1.11.0-py3-none-any.whl (201 kB)
  DEPRECATION: Wheel filename 'pytz-2013d-py3-none-any.whl' is not correctly normalised. Future versions of pip will raise the following error:
  Invalid wheel filename (invalid version): 'pytz-2013d-py3-none-any'

   pip 25.3 will enforce this behaviour change. A possible replacement is to rename the wheel to use a correctly normalised name (this may require updating the version in the project metadata). Discussion can be found at https://github.com/pypa/pip/issues/12938
  DEPRECATION: Wheel filename 'pytz-2012j-py3-none-any.whl' is not correctly normalised. Future versions of pip will raise the following error:
  Invalid wheel filename (invalid version): 'pytz-2012j-py3-none-any'

   pip 25.3 will enforce this behaviour change. A possible replacement is to rename the wheel to use a correctly normalised name (this may require updating the version in the project metadata). Discussion can be found at https://github.com/pypa/pip/issues/12938
  DEPRECATION: Wheel filename 'pytz-2012h-py3-none-any.whl' is not correctly normalised. Future versions of pip will raise the following error:
  Invalid wheel filename (invalid version): 'pytz-2012h-py3-none-any'

   pip 25.3 will enforce this behaviour change. A possible replacement is to rename the wheel to use a correctly normalised name (this may require updating the version in the project metadata). Discussion can be found at https://github.com/pypa/pip/issues/12938
  DEPRECATION: Wheel filename 'pytz-2012g-py3-none-any.whl' is not correctly normalised. Future versions of pip will raise the following error:
  Invalid wheel filename (invalid version): 'pytz-2012g-py3-none-any'

   pip 25.3 will enforce this behaviour change. A possible replacement is to rename the wheel to use a correctly normalised name (this may require updating the version in the project metadata). Discussion can be found at https://github.com/pypa/pip/issues/12938
  DEPRECATION: Wheel filename 'pytz-2012f-py3-none-any.whl' is not correctly normalised. Future versions of pip will raise the following error:
  Invalid wheel filename (invalid version): 'pytz-2012f-py3-none-any'

   pip 25.3 will enforce this behaviour change. A possible replacement is to rename the wheel to use a correctly normalised name (this may require updating the version in the project metadata). Discussion can be found at https://github.com/pypa/pip/issues/12938
  DEPRECATION: Wheel filename 'pytz-2012d-py3-none-any.whl' is not correctly normalised. Future versions of pip will raise the following error:
  Invalid wheel filename (invalid version): 'pytz-2012d-py3-none-any'

   pip 25.3 will enforce this behaviour change. A possible replacement is to rename the wheel to use a correctly normalised name (this may require updating the version in the project metadata). Discussion can be found at https://github.com/pypa/pip/issues/12938
  DEPRECATION: Wheel filename 'pytz-2011n-py3-none-any.whl' is not correctly normalised. Future versions of pip will raise the following error:
  Invalid wheel filename (invalid version): 'pytz-2011n-py3-none-any'

   pip 25.3 will enforce this behaviour change. A possible replacement is to rename the wheel to use a correctly normalised name (this may require updating the version in the project metadata). Discussion can be found at https://github.com/pypa/pip/issues/12938
  DEPRECATION: Wheel filename 'pytz-2011k-py3-none-any.whl' is not correctly normalised. Future versions of pip will raise the following error:
  Invalid wheel filename (invalid version): 'pytz-2011k-py3-none-any'

   pip 25.3 will enforce this behaviour change. A possible replacement is to rename the wheel to use a correctly normalised name (this may require updating the version in the project metadata). Discussion can be found at https://github.com/pypa/pip/issues/12938
  DEPRECATION: Wheel filename 'pytz-2011j-py3-none-any.whl' is not correctly normalised. Future versions of pip will raise the following error:
  Invalid wheel filename (invalid version): 'pytz-2011j-py3-none-any'

   pip 25.3 will enforce this behaviour change. A possible replacement is to rename the wheel to use a correctly normalised name (this may require updating the version in the project metadata). Discussion can be found at https://github.com/pypa/pip/issues/12938
  DEPRECATION: Wheel filename 'pytz-2011h-py3-none-any.whl' is not correctly normalised. Future versions of pip will raise the following error:
  Invalid wheel filename (invalid version): 'pytz-2011h-py3-none-any'

   pip 25.3 will enforce this behaviour change. A possible replacement is to rename the wheel to use a correctly normalised name (this may require updating the version in the project metadata). Discussion can be found at https://github.com/pypa/pip/issues/12938
  DEPRECATION: Wheel filename 'pytz-2011g-py3-none-any.whl' is not correctly normalised. Future versions of pip will raise the following error:
  Invalid wheel filename (invalid version): 'pytz-2011g-py3-none-any'

   pip 25.3 will enforce this behaviour change. A possible replacement is to rename the wheel to use a correctly normalised name (this may require updating the version in the project metadata). Discussion can be found at https://github.com/pypa/pip/issues/12938
  DEPRECATION: Wheel filename 'pytz-2011e-py3-none-any.whl' is not correctly normalised. Future versions of pip will raise the following error:
  Invalid wheel filename (invalid version): 'pytz-2011e-py3-none-any'

   pip 25.3 will enforce this behaviour change. A possible replacement is to rename the wheel to use a correctly normalised name (this may require updating the version in the project metadata). Discussion can be found at https://github.com/pypa/pip/issues/12938
  DEPRECATION: Wheel filename 'pytz-2011d-py3-none-any.whl' is not correctly normalised. Future versions of pip will raise the following error:
  Invalid wheel filename (invalid version): 'pytz-2011d-py3-none-any'

   pip 25.3 will enforce this behaviour change. A possible replacement is to rename the wheel to use a correctly normalised name (this may require updating the version in the project metadata). Discussion can be found at https://github.com/pypa/pip/issues/12938
Collecting pytz>=2023.3.post1 (from -r requirements.txt (line 3))
  WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ProtocolError('Connection aborted.', RemoteDisconnected('Remote end closed connection without response'))': /simple/pytz/pytz-2025.2-py3-none-any.whl
  Downloading https://www.piwheels.org/simple/pytz/pytz-2025.2-py3-none-any.whl (509 kB)
Collecting tzlocal>=5.1 (from -r requirements.txt (line 4))
  Downloading https://www.piwheels.org/simple/tzlocal/tzlocal-5.3.1-py3-none-any.whl (18 kB)
Collecting timezonefinder>=6.2.0 (from -r requirements.txt (line 5))
[... 22 lines cut ...]
        File "/home/pi/PiClock/venv/lib/python3.9/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 373, in main
          json_out["return_val"] = hook(**hook_input["kwargs"])
        File "/home/pi/PiClock/venv/lib/python3.9/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 175, in prepare_metadata_for_build_wheel
          return hook(metadata_directory, config_settings)
        File "/tmp/pip-build-env-g2lliwgx/overlay/lib/python3.9/site-packages/scikit_build_core/build/__init__.py", line 96, in prepare_metadata_for_build_wheel
          return _build_wheel_impl(
        File "/tmp/pip-build-env-g2lliwgx/overlay/lib/python3.9/site-packages/scikit_build_core/build/wheel.py", line 176, in _build_wheel_impl
          return _build_wheel_impl_impl(
        File "/tmp/pip-build-env-g2lliwgx/overlay/lib/python3.9/site-packages/scikit_build_core/build/wheel.py", line 248, in _build_wheel_impl_impl
[... cut ...]
note: This is an issue with the package mentioned above, not pip.
hint: See above for details.

Ive searched and googled as far as I am able to try and resolve the issue. Ive tried just moving ahead but I get all sorts of missing modules when starting. Im assuming its a compiling issue or something with cmake but I dont know what to do. Ive tried several times to move up to bookworm thinking that might be the issue but sadly this raspberry pi just cant run bookworm. If thats the case I guess its end of life for this RPI.

GrampDamnpus, 2025-09-26:
File "/tmp/pip-build-env-g2lliwgx/overlay/lib/python3.9/site-packages/scikit_build_core/cmake.py", line 67, in default_search
raise CMakeNotFoundError(msg)
scikit_build_core.errors.CMakeNotFoundError: Could not find CMake with version >=3.15
[end of output]

Using this I figured Id try installing cmake. Im not getting a new message that says the cmake version is outdated...at 3.18. So Im currently trying to update cmake.

SerBrynden, 2025-09-26:
I'd avoid going down the rabbit hole of installing things manually. What Raspberry Pi board do you have? You should be able to install Bookworm on all boards. If you can't do a fresh install of Bookworm, you'll need to try a different board. It's not going to work with Bullseye.